mph-/lcapy #3: Can't run simple example

MuriloHMoreira, 2016-10-20:
Hello there,
I'm sorry to bother, but I'm afraid that I'm not the only one who would face this problem.
I have little experience with programming but really liked this project, so I installed but can't run the simple example present in the documentation.

When I run this on a Jupyter Notebook (Python 3.5): 
```
from lcapy import *
from numpy import logspace
from matplotlib.pyplot import savefig, show

N = R(10) + C(1e-4) + L(1e-3)

vf = logspace(0, 5, 400)
Z = N.Z.frequency_response().magnitude.plot(vf,0)

show()
```

I get 

> NameError                                 Traceback (most recent call last)
> <ipython-input-12-de33b3214e6d> in <module>()
>       6 
>       7 vf = logspace(0, 5, 400)
> ----> 8 Z = N.Z.frequency_response().magnitude.plot(vf,0)
>       9 
>      10 show()
> 
> C:\Anaconda3\lib\site-packages\lcapy\core.py in frequency_response(self, fvector)
>    1355         """
>    1356 
> -> 1357         X = self(j * 2 * pi * f)
>    1358 
>    1359         if fvector is None:
> 
> C:\Anaconda3\lib\site-packages\lcapy\core.py in __call__(self, arg)
>     850             return np.array([self._subs1(self.var, arg1) for arg1 in arg])
>     851 
> --> 852         return self._subs1(self.var, arg)
>     853 
>     854     def subs(self, *args, **kwargs):
> 
> C:\Anaconda3\lib\site-packages\lcapy\core.py in _subs1(self, old, new, **kwargs)
>     834         old = context.symbols[name]
>     835 
> --> 836         return cls(self.expr.subs(old, expr))
>     837 
>     838     def __call__(self, arg):
> 
> C:\Anaconda3\lib\site-packages\lcapy\core.py in __init__(self, val)
>    2131     def __init__(self, val):
>    2132 
> -> 2133         super(Zf, self).__init__(val, **assumptions)
>    2134         self._fourier_conjugate_class = Zt
>    2135 
> 
> NameError: name 'assumptions' is not defined

I also tried to plot

```
from lcapy import *
from numpy import linspace
from matplotlib.pyplot import savefig, show

N = Vdc(20) + R(10) + C(1e-4)

tv = linspace(0, 0.01, 1000)
N.Isc.transient_response().plot(tv)

show()

```

 but ran into the problems:

> TypeError                                 Traceback (most recent call last)
> C:\Anaconda3\lib\site-packages\lcapy\core.py in evaluate(self, arg)
>     756         try:
> --> 757             result = func(v1)
>     758             response = complex(result)
> 
> C:\Anaconda3\lib\site-packages\numpy\__init__.py in <lambda>(_Dummy_99)
> 
> C:\Anaconda3\lib\site-packages\sympy\core\relational.py in __nonzero__(self)
>     194     def __nonzero__(self):
> --> 195         raise TypeError("cannot determine truth value of Relational")
>     196 
> 
> TypeError: cannot determine truth value of Relational
> 
> During handling of the above exception, another exception occurred:
> 
> RuntimeError                              Traceback (most recent call last)
> <ipython-input-16-354efcde2706> in <module>()
>       6 
>       7 tv = linspace(0, 0.01, 1000)
> ----> 8 N.Isc.transient_response().plot(tv)
>       9 
>      10 show()
> 
> C:\Anaconda3\lib\site-packages\lcapy\core.py in plot(self, t, **kwargs)
>    1619 
>    1620         from lcapy.plot import plot_time
> -> 1621         plot_time(self, t, **kwargs)
>    1622 
>    1623     def canonical(self):
> 
> C:\Anaconda3\lib\site-packages\lcapy\plot.py in plot_time(obj, t, **kwargs)
>     146         t = np.linspace(t[0], t[1], 400)
>     147 
> --> 148     v = obj.evaluate(t)
>     149 
>     150     ax = kwargs.pop('axes', None)
> 
> C:\Anaconda3\lib\site-packages\lcapy\core.py in evaluate(self, arg)
>     763                 raise RuntimeError(
>     764                     'Cannot evaluate expression %s,'
> --> 765                     ' due to undetermined conditional result' % self)
>     766 
>     767             raise RuntimeError(
> 
> RuntimeError: Cannot evaluate expression Piecewise((-exp(-1000*t)/(Heaviside(0) - 1), t >= 0)), due to undetermined conditional result
> 
> In [ ]:
> 

​sorry for wasting your time, I just want to help and be helped.
Thanks in advance,
Best regards, Murilo Moreira.

mph-, 2016-10-25:
On Thu, 2016-10-20 at 10:45 -0700, Murilo Henrique Moreira wrote:

Hello Murilo

> I also tried to plot
> 
> from lcapy import *
> from numpy import linspace
> from matplotlib.pyplot import savefig, show
> 
> N = Vdc(20) + R(10) + C(1e-4)
> 
> tv = linspace(0, 0.01, 1000)
> N.Isc.transient_response().plot(tv)
> 
> show()

I fixed this problem that was due to a change in the behaviour of SymPy.
For this example Vdc should be Vstep otherwise a dc analysis will be
performed.  I've corrected the tutorial.

Michael.

mph-, 2016-10-25:
Thanks for letting me know Murilo.  I've pushed a patch to fix a newly
introduced bug.

BTW, the plot method only has a single argument.  The others need to be
named keyword arguments.  So

Z = N.Z.frequency_response().magnitude.plot(vf,0)

should be

Z = N.Z.frequency_response().magnitude.plot(vf)

I cannot replicate your second problem.  Which version of sympy are you
using?

Michael.

MuriloHMoreira, 2016-10-27:
Hey Michael, thanks for the feedback,
I would like to ask you witch way is better to update this package.
Should I download the zip again, delete the old folder and uninstall the old version?
By the way, I'm using sympy v.1.0
Best regards, Murilo.

mph-, 2016-10-30:
Hello Murilo

You can either download a new .zip file or better still clone the
repository using git (you won't regret learning how to use git).  There
is no need to un-install. Just run python setup.py install   

Michael.

closed 2016-11-04 by mph-