mph-/lcapy #4: getting Heavisid() functions when calculating impedance

eriic1, 2016-10-21:
Hello,

When I follow along in the tutorial at on http://lcapy.elec.canterbury.ac.nz/tutorial.html
I am getting unexpected results.  For example 

>>>from lcapy import *
>>> N = R(5) | L(20) | C(10)
>>> N
R(5) | L(20) | C(10)
>>> N.Z

I get 

  ⎛  Heaviside(0)   1⎞
s⋅⎜- ──────────── + ─⎟
  ⎝       5         5⎠
──────────────────────
     2   s     1
    s  + ── + ───
         50   200

instead of

      20⋅s
────────────────
     2
200⋅s  + 4⋅s + 1

All the impedance calculations have the Heaviside() function in them. 

I am not sure if it is because there is something wrong with my setup but I can provide more examples if needed if it will help in determining what is going wrong.

I am using:
Python version 3.5,2
IPython version 5.1.0
numpy version 1.11.2
sympy versions 1.0-1

Thank you for any help.

Eric

eriic1, 2016-10-22:
My aplogies to Oliver for the mispelling of Heaviside. :-)

Your last commits solved the problem and now I can correctly get the impedance using the examples in the tutorial.  

Thank you very much,

Eric

closed 2016-10-22 by eriic1

mph-, 2016-10-25:
Thanks for the info.

It looks like something has changed with sympy.  I'll upgrade and see
what happened.

Michael.

mph-, 2016-10-25:
Hello Eric

I've pushed a fix for this problem.  SymPy was punting on evaluating
Heaviside(0) whereas older versions simplified this to 0.5.

Michael.