mph-/lcapy #6: Difficulty Running Notebook Examples

kohaugustine, 2017-01-15:
Hi, I have just installed lcapy on my machine. The way I installed was slightly different from what is suggested in the docs because I use conda for python environment management. The main differences were that I created a new conda environment, installed sympy, numpy, ipython and matplotlib within it, then after activating that environment, I used `pip install -e /path/to/lcapy/on/my/machine` to install the lcapy as a package within my conda environment. So I did not run the setup.py and used the pip install instead.

This seemed to work perfectly fine. I was able to start importing lcapy within active ipython sessions. Then I tried running the python notebook examples and that's where the problem started. For example, when I run the RC-lpf1-buffer-loaded.ipynb, the first cell works fine to generate the  drawing the circuit upon the `cct.draw()` command. But the next cell that performs `H = cct.RL.V / cct.Vi.V` is where the issue starts. Hitting that line, I get the error `TypeError: Unsupported types for /: 'Vsuper' and 'Vsuper'`. 

What is the reason for this bug, and is it related to the way I have installed the package? I doubt it would be though, because the problem of the error appears to be that the `/` operator was not defined to divide Vsuper over Vsuper. Am I missing something here?

I'm starting on my very first analog signals and systems class within a couple of days and I really hope to get lcapy working on my machine to enrich my learning. Thank you so much for creating this library and I hope you can help me with this issue.

eriic1, 2017-01-16:
I do not believe it has anything to do with your install.  I just updated with the latest on github and get the same error when running the notebook with python3 using the notebook from the doc/examples/notebooks directory.

kohaugustine, 2017-01-16:
I am likewise using python 3 environment. Since you mentioned using a python3 environment, just now I decided to give it a try under python2. I created a new python 2 environment, but running the notebooks under this also gave the same errors. So it looks like it isn't about the python versions...

psychemedia, 2017-12-08:
I am running `lcapy` under python3 and whilst I can render circuits, I get errors whenever I try a calculation. For example:

```python
from lcapy import Circuit, j, omega
cct = Circuit()
cct.add("""
Vi 1 0_1; down
C 1 2; right, size=1.5
R 2 0; down
W 0_1 0; right
W 0 0_2; right, size=0.5
P1 2_2 0_2; down
W 2 2_2;right, size=0.5""")

H = (cct.R.V / cct.Vi.V).simplify()

TypeError 

Traceback (most recent call last)
<ipython-input-9-6806f729093f> in <module>()
----> 1 H = (cct.R.V / cct.Vi.V).simplify()

/srv/venv/lib/python3.5/site-packages/lcapy/core.py in __truediv__(self, x)
   3153 
   3154     def __truediv__(self, x):
-> 3155         return self.__div__(x)
   3156 
   3157     def cpt(self):

/srv/venv/lib/python3.5/site-packages/lcapy/core.py in __div__(self, x)
   3145             raise TypeError('Cannot divide types %s and %s. '
   3146             'You need to extract a specific component, e.g., a.s / b.s' %
-> 3147             (type(self).__name__, type(x).__name__))
   3148 
   3149         if not isinstance(x, Zs):

TypeError: Cannot divide types Vsuper and Vsuper. You need to extract a specific component, e.g., a.s / b.s
```

eriic1, 2017-12-15:
Tony,

If you want to find the simplified term for H you can do something like 
this:

H = (cct.R.V('s') / cct.Vi.V('s').simplify()

It looks like each of the terms are a dictionary with a key of 's' and 
the value is the equation you are looking for.  This will let you get 
the value of each term and use them.

I get using ipython3 with Ubuntu 16.04 the following:

cct = Circuit()
cct.add("""
Vi 1 0_1 step 20; down
C 1 2; right, size=1.5
R 2 0; down
W 0_1 0; right
W 0 0_2; right, size=0.5
P1 2_2 0_2; down
W 2 2_2;right, size=0.5""")

H = (cct.R.V('s') / cct.Vi.V('s')).simplify()
H

  C⋅R⋅s
─────────
C⋅R⋅s + 1

Hope this helps you,

Eric

psychemedia, 2017-12-15:
Hi Eric

Thanks - I'd missed the `step 20` setting... oops... (too much haste in producing quick demos.)

--tony