mph-/lcapy #8: Trouble getting lcapy to work on openSUSE due to circuitikz

eriic1, 2018-01-28:
Hello,

I am trying to get lcapy working on an openSUSE 42.3 installation.  I believe I have all the dependencies required but run into an error when running  a simple file such as:

from lcapy import Circuit, s
cct = Circuit()
cct.add('V1 1 0 {u(t)}; down')
cct.add('R1 1 2 4; right=1.5')
cct.add('C1 2 3 0.1; down=1.5')
cct.add('W1 0 3; right')
cct.draw()

This returns:

RuntimeError: circuitikz is not installed

I know circuitikz is installed and can use it in a latex document.   

Do you have any suggestions on how to get this working or can you help me in finding out how lcapy determines the status of circuitikz?  I noticed that openSUSE and Ubuntu install their texlive sty files in different places and wonder if that could be the issue.

Thank you for any help.

Eric

mph-, 2018-01-30:
Hello Eric

What happens when you run pdflatex on this file?

\documentclass[a4paper]{standalone}
\usepackage{circuitikz}
\begin{document}
\end{document}

Michael

eriic1, 2018-02-09:
Thank you Michael,

I got the error with pdflatex below:

! LaTeX Error: File `standalone.cls' not found.

So I installed texlive-standalone and now everything is working great.

I was using this as my test document with pdflatex, and it did not give me any errors.

\documentclass{article}
\usepackage{mathtools}
\usepackage{graphicx}
\usepackage{circuitikz}
\DeclareGraphicsExtensions{.pdf,.png,.jpg}
\title{}
\author{}
\date{}
\begin{document}
\maketitle
\begin{circuitikz}[scale =1.2]
\draw (0,0)
            to[V,v=$U_q$] (0,2)
            to[short] (2,2)
            to[R=$R_1$] (2,0)
            to[short] (0,0);
\end{circuitikz}
\end{document}

and then I noticed the {standalone} on your test document. 

Thank you very much,

Eric

closed 2018-02-09 by eriic1