mph-/lcapy #10: new sympy gives a deprecation warning

peendebak, 2018-07-09:
Minimal example
``` python
from lcapy import Circuit # , s, t, f, R, L, C
cct = Circuit()
cct.add('V1 1 0 {u(t)}; down')
cct.add('L 1 2 1; right=2')
cct.add('C 2 3 1e-6; down=1.5')
cct.add('W24 2 4 ; right=2')
cct.add('R 4 5 ; down=1.5, v=$\Omega$')
cct.add('W1 0 3; right')
cct.add('W35 3 5; right')

cct.draw()
```
gives the following warning
```
c:\develop\anaconda3\lib\site-packages\sympy\core\decorators.py:38: SymPyDeprecationWarning: 

simplify has been deprecated since SymPy 1.1. Use
expand(diracdelta=True, wrt=x) instead. See
https://github.com/sympy/sympy/issues/12859 for more info.

  _warn_deprecation(wrapped, 3)
```

psychemedia, 2018-08-07:
`sympy` version 1.2 seems to break things altogether.

```
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-8-1f80ecb614d7> in <module>()
----> 1 from lcapy import Circuit, j, omega
 
/srv/conda/lib/python3.6/site-packages/lcapy/__init__.py in <module>()
     35 # Add the modules that get searched to allow 'from lcapy import V'
     36 # rather then having to specify module, 'from lcapy.oneport import V'
---> 37 from . import core
     38 __all__.extend(core.__all__)
     39 from .core import *

/srv/conda/lib/python3.6/site-packages/lcapy/core.py in <module>()
     13 from __future__ import division
     14 from lcapy.latex import latex_str
---> 15 from lcapy.acdc import is_dc, is_ac, is_causal, ACChecker
     16 from lcapy.sympify import canonical_name, sympify1, symbols_find
     17 from lcapy.ratfun import Ratfun, _zp2tf

/srv/conda/lib/python3.6/site-packages/lcapy/acdc.py in <module>()
----> 1 from lcapy.sympify import sympify1
      2 import sympy as sym
      3 from sympy import symbols, I, exp, cos, pi, sin, atan2, sqrt
      4 
      5 class CausalChecker(object):

/srv/conda/lib/python3.6/site-packages/lcapy/sympify.py in <module>()
      1 from sympy.parsing.sympy_parser import parse_expr, auto_number, rationalize
----> 2 from sympy.parsing.sympy_tokenize import NUMBER, STRING, NAME, OP
      3 from sympy import Basic, Symbol, Expr
      4 import sympy as sym
      5 import re

ModuleNotFoundError: No module named 'sympy.parsing.sympy_tokenize'
```

It __is__ in the [docs for `sympy v1.1`](http://docs.sympy.org/1.1/modules/parsing.html#parsing-functions-reference) but __not__ for `sympy v1.2+`](http://docs.sympy.org/1.2/modules/parsing.html#parsing-functions-reference).

mph-, 2019-01-02:
This is  a problem with sympy.  It really needs fixing in sympy but I'll try to come up with a workaround to paper over it.

Michael.

psychemedia, 2019-01-02:
It looks like `sympy_tokenize`, as called by `lcapy`, was removed some time ago? https://github.com/sympy/sympy/pull/14085

mph-, 2019-01-02:
@psychemedia which version of Lcapy are you using?

On perusal, I found that I fixed this on 2018-08-19

psychemedia, 2019-01-03:
@mph- An older version probably! I just picked up on this again from a Github alert on the repo.

Thanks for the patches.

mph-, 2019-01-03:
Okay. I'll close this then.

closed 2019-01-03 by mph-