mph-/lcapy #14: Little issues when running examples

bcornelusse, 2018-12-05:
Hello, 

I am considering using lcapy for my course at University of Liège, Belgium. Lcapy looks very nice. 
So I started browsing and running a few experiments. Many things work (e.g plotting networks, some impedance magnitude plots, etc.), but I have also little issues: 

I am using commit 785e12e6740eba0a6901e6d7e35c6f62b981be6f. I run it on a mac in a virtual environment based on python 3.6.6, with the following installed packages: [requirements.txt](https://github.com/mph-/lcapy/files/2648058/requirements.txt)

E.g. when I run 
```
$ python demo/oneport/demo1.py 
Traceback (most recent call last):
  File "demo/oneport/demo1.py", line 11, in <module>
    ax.plot(t, a.V.impulse_response(t), linewidth=2)
  File "/Users/bcr/Documents/ws/scripts_ELEC0053/cenv/lib/python3.6/site-packages/lcapy/oneport.py", line 117, in V
    return self.Voc()
  File "/Users/bcr/Documents/ws/scripts_ELEC0053/cenv/lib/python3.6/site-packages/lcapy/oneport.py", line 519, in Voc
    return self.cct.Voc(1, 0)
  File "/Users/bcr/Documents/ws/scripts_ELEC0053/cenv/lib/python3.6/site-packages/lcapy/netlist.py", line 410, in Voc
    return self.get_Vd(Np, Nm)
  File "/Users/bcr/Documents/ws/scripts_ELEC0053/cenv/lib/python3.6/site-packages/lcapy/netlist.py", line 1171, in get_Vd
    Vd = sub.get_Vd(Np, Nm)
  File "/Users/bcr/Documents/ws/scripts_ELEC0053/cenv/lib/python3.6/site-packages/lcapy/netlist.py", line 1211, in get_Vd
    self._solve()
  File "/Users/bcr/Documents/ws/scripts_ELEC0053/cenv/lib/python3.6/site-packages/lcapy/mna.py", line 147, in _solve
    self._analyse()
  File "/Users/bcr/Documents/ws/scripts_ELEC0053/cenv/lib/python3.6/site-packages/lcapy/mna.py", line 122, in _analyse
    num_nodes = len(self.node_list) - 1
  File "/Users/bcr/Documents/ws/scripts_ELEC0053/cenv/lib/python3.6/site-packages/lcapy/mna.py", line 74, in node_list
    node_list.insert(0, node_list.pop(node_list.index('0')))
ValueError: '0' is not in list
```

Also I have 
```
$ python demo/oneport/series-RLC-frequency1.py 
/Users/bcr/Documents/ws/scripts_ELEC0053/cenv/lib/python3.6/site-packages/matplotlib/ticker.py:2241: UserWarning: Data has no positive values, and therefore cannot be log-scaled.
  "Data has no positive values, and therefore cannot be "
```
nothing shows up on the plot.

And 
```
python demo/oneport/series-RLC.py 
```
runs but the plot is flat (current = 0 for all time steps.)

Can you maybe give a hint? Is my configuration not supported?

Thanks,
All the best, 
Bertrand.

eriic1, 2018-12-29:
Have you tried with the latest sources.  Yours are over a month old.  I tried using an updated Ubuntu 16.04 with the latest source code and the commands give what I think are the expected outputs in the plots.  

I would not think this is caused by using a mac or the virtual environment would cause the problems as other plots are working correct for you.

mph-, 2019-01-02:
Hi, thanks for letting me know.  These demos were horribly out of date and lurk from a previous era before I added netlist support.

I have fixed them up.

Michael

closed 2019-01-02 by mph-