mph-/lcapy #17: Op-Amps Unsupported?

firebolt55439, 2019-03-30:
None of the op-amp examples work with the newest version of `lcapy` since the `Opamp` class is no longer exposed for imports. Is op-amp analysis supported for simple op-amp networks? 

Not looking for anything too fancy. I keep getting "MNA A matrix not invertible" errors when attempting to analyze voltage gain across simple op-amp configurations.

mph-, 2019-03-31:
Hello

Opamps should be working.  There is an example in the `doc/examples/notebooks/opamp-noninverting-amplifier1.ipynb` notebook.

The matrix not invertible error usually occurs when you do not have a DC path for a node to ground.

Could you post a simplified testcase please?

Michael.

firebolt55439, 2019-03-31:
Ah I see, I was looking at the old examples, which is why I couldn't get anything to work. No wonder. Thanks for your help :)

closed 2019-03-31 by firebolt55439

mph-, 2019-03-31:
Ah, which old examples did you find?   If they are in the docs, I'll remove or update them.

firebolt55439, 2019-03-31:
Pretty much all the ones in here: [https://github.com/mph-/lcapy/tree/master/demo/notebooks](https://github.com/mph-/lcapy/tree/master/demo/notebooks)