mph-/lcapy #18: Not lowcase filenames on Windows problem

algo99, 2019-04-12:
Hi,

first I want to thank you for this amazing package.

Trying to use it on Windows I noted that there are two example files (may be erroneously committed) which names differ only in case:

lcapy/doc/examples/networks/series-VRC1-**I**sc.py
lcapy/doc/examples/networks/series-VRC1-**i**sc.py

Unfortunately, it looks like git-on-windows can not cope with this properly.
And after `git clone ...` i get immediately:
```
e:\Users\xyz\workplace\git\lcapy (master -> origin)
λ git status                                                               
On branch master                                                           
Your branch is up to date with 'origin/master'.                            
                                                                           
Changes not staged for commit:                                             
  (use "git add <file>..." to update what will be committed)               
  (use "git checkout -- <file>..." to discard changes in working directory)
                                                                           
        modified:   doc/examples/networks/series-VRC1-Isc.py               
                                                                           
no changes added to commit (use "git add" and/or "git commit -a")          
```

Do you have any plan to release/update this package on PyPi as well ?

mph-, 2019-04-13:
Thanks for you kind words.

Ah Windows filenames, what a nuisance!  I will try to think up some new names.

I've had a crack at making things work for PyPi and have pushed to the test server.  Would you mind trying https://test.pypi.org/project/lcapy/ ?

mph-, 2019-04-14:
I have renamed the files.  Would you mind confirming if this fixes the problem?

algo99, 2019-04-14:
Well, it works as far as I can see. The functionality we use at the moment:
- create a netlist and draw it
- get an impedance of a network, apply subs, and evaluate in Laplace domain

Few things I would wish to see change is summarized in the attached patch of `setup.py` [setup_py.zip](https://github.com/mph-/lcapy/files/3077217/setup_py.zip):
- usage of `install_requires` instead of `requires` to force pip to install missing packages
- adding `matplotlib` to the list of required dependencies
- usage of `entry_points` in preference to the `scripts` keyword to allow pip to create the appropriate form of executable for the target platform (for this to work, I think you should move schtex.py one directory level up in your source distribution)

The reason of the third point above is, that in this case pip creates an executable shim file that is put into the "bin" directory (`.virtualenv/<myenv>/bin/schtex` on Linux, `<PythonRoot>/Scripts/schtex.exe` on Windows ) and put `schtex.py` into `site-packages` directory so when `schetex` will be run in a console `schetex.py` will be correctly found and imported by Python.

algo99, 2019-04-14:
> I have renamed the files. Would you mind confirming if this fixes the problem?

Now OK, thank you!

closed 2019-04-15 by algo99