mph-/lcapy #19: Please consider publish / update lcapy package on PyPi

algo99, 2019-04-15:
> I've had a crack at making things work for PyPi and have pushed to the test server. Would you mind trying https://test.pypi.org/project/lcapy/ ?

Well, it works as far as I can see. The functionality we use at the moment:
- create a netlist and draw it
- get an impedance of a network, apply subs, and evaluate in Laplace domain

Few things I would wish to see change is summarized in the attached patch of `setup.py` [setup_py.zip](https://github.com/mph-/lcapy/files/3077217/setup_py.zip):
- usage of `install_requires` instead of `requires` to force pip to install missing packages
- adding `matplotlib` to the list of required dependencies
- usage of `entry_points` in preference to the `scripts` keyword to allow pip to create the appropriate form of executable for the target platform (for this to work, I think you should move schtex.py one directory level up in your source distribution)

The reason of the third point above is, that in this case pip creates an executable shim file that is put into the "bin" directory (`.virtualenv/<myenv>/bin/schtex` on Linux, `<PythonRoot>/Scripts/schtex.exe` on Windows ) and put `schtex.py` into `site-packages` directory so when `schetex` will be run in a console `schetex.py` will be correctly found and imported by Python.

mph-, 2019-04-16:
Many thanks for your suggestions.   I had added install_requires but was blind to the old requires...

Thanks for figuring out the entry_point stuff.   I've uploaded your suggestions to  https://test.pypi.org/project/lcapy/

algo99, 2019-04-17:
Not sure why, but if I try to install (with WinPython) lcapy from test PyPi and sympy is not installed I get
```
λ pip install -i https://test.pypi.org/simple/ lcapy                                                                                               
Looking in indexes: https://test.pypi.org/simple/                                                                                                  
Collecting lcapy                                                                                                                                   
  Downloading https://test-files.pythonhosted.org/packages/b5/1c/9a8b23794622822a890009627f6cd0c709a2b5acad9782b2368ce32478e4/lcapy-0.33.0-py3-none
-any.whl (152kB)                                                                                                                                   
    100% |████████████████████████████████| 153kB 3.9MB/s                                                                                          
Requirement already satisfied: scipy>=1.2.0 in c:\3rd\winpython0\python-3.7.2.amd64\lib\site-packages (from lcapy) (1.2.1)                         
Requirement already satisfied: numpy>=1.12 in c:\3rd\winpython0\python-3.7.2.amd64\lib\site-packages (from lcapy) (1.16.2+mkl)                     
Collecting sympy>=1.3 (from lcapy)                                                                                                                 
  Could not find a version that satisfies the requirement sympy>=1.3 (from lcapy) (from versions: )                                                
No matching distribution found for sympy>=1.3 (from lcapy)                                                                                         
```

Using git version of lcapy and building the wheel package by myself and then installing it goes right through 
```
e:\Users\zkr\workplace\git\lcapy (master -> origin)
λ python setup.py bdist_wheel
...
e:\Users\zkr\workplace\git
λ pip install -U lcapy\dist\lcapy-0.33.1-py3-none-any.whl
Processing e:\users\zkr\workplace\git\lcapy\dist\lcapy-0.33.1-py3-none-any.whl
Requirement already satisfied, skipping upgrade: scipy>=1.2.0 in c:\3rd\winpython0\python-3.7.2.amd64\lib\site-packages (from lcapy==0.33.1) (1.2.1)
Collecting sympy>=1.3 (from lcapy==0.33.1)
  Using cached https://files.pythonhosted.org/packages/21/21/f4105795ca7f35c541d82c5b06be684dd2f5cb4f508fb487cd7aea4de776/sympy-1.4-py2.py3-none-any.whl
Requirement already satisfied, skipping upgrade: matplotlib>=2.2.2 in c:\3rd\winpython0\python-3.7.2.amd64\lib\site-packages (from lcapy==0.33.1) (3.0.3)
Requirement already satisfied, skipping upgrade: numpy>=1.12 in c:\3rd\winpython0\python-3.7.2.amd64\lib\site-packages (from lcapy==0.33.1) (1.16.2+mkl)
Requirement already satisfied, skipping upgrade: mpmath>=0.19 in c:\3rd\winpython0\python-3.7.2.amd64\lib\site-packages (from sympy>=1.3->lcapy==0.33.1) (1.1.0)
Requirement already satisfied, skipping upgrade: python-dateutil>=2.1 in c:\3rd\winpython0\python-3.7.2.amd64\lib\site-packages (from matplotlib>=2.2.2->lcapy==0.33.1) (2.8.0)
Requirement already satisfied, skipping upgrade: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in c:\3rd\winpython0\python-3.7.2.amd64\lib\site-packages (from matplotlib>=2.2.2->lcapy==0.33.1) (2.4.0)
Requirement already satisfied, skipping upgrade: cycler>=0.10 in c:\3rd\winpython0\python-3.7.2.amd64\lib\site-packages (from matplotlib>=2.2.2->lcapy==0.33.1) (0.10.0)
Requirement already satisfied, skipping upgrade: kiwisolver>=1.0.1 in c:\3rd\winpython0\python-3.7.2.amd64\lib\site-packages (from matplotlib>=2.2.2->lcapy==0.33.1) (1.0.1)
Requirement already satisfied, skipping upgrade: six>=1.5 in c:\3rd\winpython0\python-3.7.2.amd64\lib\site-packages (from python-dateutil>=2.1->matplotlib>=2.2.2->lcapy==0.33.1) (1.12.0)
Requirement already satisfied, skipping upgrade: setuptools in c:\3rd\winpython0\python-3.7.2.amd64\lib\site-packages (from kiwisolver>=1.0.1->matplotlib>=2.2.2->lcapy==0.33.1) (41.0.0)
Installing collected packages: sympy, lcapy
Successfully installed lcapy-0.33.1 sympy-1.4
```
Is it 0.33.1 (not published yet) vs. 0.33.0 that make the difference ?

mph-, 2019-04-17:
I think the problem is that test.pypi only has old versions of scipy etc.  I will create an account on pypi and hopefully everything will work :-)

mph-, 2019-04-17:
pip3 install lcapy seems to work for me now :-)

Thanks for your help.

closed 2019-04-17 by mph-