mph-/lcapy #20: Binding symbols in state space model does not work (or is not documented)

jbulow, 2019-08-20:
Is there an example of how to bind symbols in a state space model of a circuit?

```
cct = Circuit("""
C1 1 0; down=1.5
W 1 1_0; right=2.0
W 0 0_3; right
I1 1_0 0_3; down
R1 1_0 2 ; right=1.5
W 0_3 0_5; right
C2 2 0_5 ; down
R2 2 3 ; right=1.5
W 0_5 0_7; right=1.5
V 3 0_7 ; down
; label_nodes=alpha
""")

ss=cct.ss

se = ss.state_equations()

ss.A.free_symbols

e2 = ss.A.subs(C1, 50400.0).subs(R1, 13.65).subs(R2, 1.35).subs(C2, 5613300.0)

e2 <--- symbols still unbound

e2.evalf() <---- does not eval using symbol values.

```

mph-, 2019-08-21:
It looks like you have not defined C1, R1, etc.

Try this instead...

e2 = ss.A.subs('C1', 50400.0).subs('R1', 13.65).subs('R2', 1.35).subs('C2', 5613300.0)

M~

closed 2019-08-21 by mph-

jbulow, 2019-08-21:
Sorry, I missed that line above. I did define the symbols before create cct as:

C1, R1, R2, C2 = sp.symbols('C1 R1 R2 C2')

Even when quoting the symbols the substitution does not work.

jbulow, 2019-08-21:
Found the error. I misinterpreted http://lcapy.elec.canterbury.ac.nz/expressions.html#assumptions-for-symbols : 

"By default, the symbol and expr functions assume positive=True unless real=True or positive=False are specified."

as positive=True is default for symbols. It's not. If I explicitly add positive=True when defining the symbols it works.

mph-, 2019-08-22:
Oh, this is a gnarly issue and has caused all sorts of problems that I have had to implement workarounds for.  The fundamental problem is that sympy allows multiple symbols with the same name but different assumptions. Lcapy tries to avoid this using the first assumptions provided when a symbol is created.  I would welcome any suggestions of how to make this more robust.

Another solution is to only create symbols using Lcapys' symbol function, e.g.,
C = symbol('C')

jbulow, 2019-08-22:
From a programming perspective it would make sense to refer to symbols via the circuit object as a namespace. I.e:

e2 = ss.A.subs(cct.C1, 50400.0).subs(cct.R1, 13.65).subs(cct.R2, 1.35).subs(cct.C2, 5613300.0)

This line does not give an error but it does not work as expected. 

Another alternative is to make the symbol accessible via subscription syntax:

e2 = ss.A.subs(cct['C1'], 50400.0).subs(cct['R1'], 13.65).subs(cct['R2'], 1.35).subs(cct['C2'], 5613300.0)

mph-, 2019-08-22:
O.n first thought this seems like an excellent suggestion.  Indeed, all the infrastructure is in place to support symbol contexts.  The problem is that R1 can refer to component R1 and a symbol R1.  For example,   cct.R1.i returns the current through R1.  Similarly cct['R1'] returns a node called R1.  Admittedly, this is less likely and so may be the better way forward.

Perhaps cct.symbols.R1 is the least ambiguous approach?

jbulow, 2019-08-22:
cct.symbols.R1 looks nice and improves code readability and is less ambiguous.

mph-, 2019-08-23:
I've pushed the required machinations to git but have yet to document them.

cct.symbols.R1  should now work.  Note, this only includes symbols defined by the circuit.

I also added cct.all_symbols that also includes globally defined symbols such as t and s.