mph-/lcapy #21: Formatting of circuit and equations in notebooks

jbulow, 2019-08-23:
The following example shows some issues when rendering equations and circuits.

`
import lcapy as lca
cct = lca.Circuit("""
Cabc T123 0; down=1.5
W T123 1_0; right=2.0
W 0 0_3; right
Iabc 1_0 0_3; down
Rdef 1_0 T456 ; right=1.5
W 0_3 0_5; right
Cxyz T456 0_5 ; down
Rghi T456 To ; right=1.5
W 0_5 0_7; right=1.5
Vdef To 0_7 ; down
""")

cct.draw()

ss=cct.ss
se=ss.state_equations()
se

ss.x

ss.y
`

It would be nice if node names were subscripted in the same way as component names.

Subscriptions in equations are not correctly rendered. Only first character is subscripted.

mph-, 2019-08-24:
I've improved the formatting of node names.  You will have to name the node T_123 instead of T123 since there is no simple generic way to determine the subscript.

mph-, 2019-08-25:
I've had a go at the other problems.  These are due to sympy and probably should be considered bugs.  For example:

import sympy as sym;
sym.sympify('v1_0') gives v₁ ₀
sym.latex(sym.sympify('v1_0')) gives 'v_{1 0}'

The second problem is gnarlier and is arguably a bug with sympy.
For example, sym.latex(sym.sympify('v_C_abc')) gives  'V_{C abc}'

jbulow, 2019-08-26:
Are these problems reported upstream to sympy?

mph-, 2019-08-27:
No, not by me anyway.   I'm busy for the next week so if you could do this it would be great.

jbulow, 2019-08-29:
I will look into and see if I can create a upstream issue. I'm a bit new to
both python and all the sci-packages but I think I know enough to create a
issue report now from your description of the underlying the problem.

mph-, 2019-08-29:
I've pushed a quick workaround to handle the double subscript problem.

oscargus, 2019-08-29:
This is caused by the SymPy variable naming conventions. Superscripts and subscripts are "collected" rather than "stacked" and trailing numbers are considered to be subscripts. One reason is that it should be possible to output reasonably well in other formats than LaTeX, another is to avoid complicated parsing of the variable names.

It is possible to give explicit LaTeX names (any name with a `{` in it won't be touched), but one will have to live with ugly names in other formats.

See: https://github.com/sympy/sympy/blob/master/sympy/printing/conventions.py

oscargus, 2019-08-29:
I have not looked into the code here (looks interesting though), but creating strings to be sympified (if that is what you are doing, I get that impression), is probably not the recommended approach. Much better to try to create the expressions directly, which is most likely considerably faster as well and use `Symbol` to get proper names etc (if multiple sub-scripts levels etc is required).

mph-, 2019-08-29:
Ah, I changed the syntax for hidden nodes but have not documented it yet while trying to see what unforeseen side-effects occur.   If a node name starts with an underscore it is hidden or if it starts with a digit and has an underscore it is hidden.  So T_123 is not hidden but _T_123 and 0_1 are hidden.

With regards to double subscripts such as v_T_123, I format this for unicode as v_T123 and for LaTeX as v_{T_{123}}.    I'm surprised SymPy does not do this.

oscargus, 2019-09-30:
How would you format v_T^2 vs v^2_T? I think that is really what it boils down to. Add arbitrary number of sub- and super-scripts if you think the first one is easy...