mph-/lcapy #25: Rendering missing wires when label_node=True...

frohro, 2019-12-08:
Hi Mark,
When I use label_node=True (something I find very very handy for debugging my mistake when I make up a circuit), I find the rendering of some of the wires in this example are not there.

`cct2 = Circuit("""
V1 1 6 8; down
R1 1 7 3; right=2
R2 7 3 1; right
R3 6 0 5; right
R7 3 0_0 5; down
R4 1 4 4;up
V2 4 5 6; right=4, i=I_a
R5 5 3; down
W 0 0_0; right=2
I1 7 0 2; down
""")
cct2.draw(label_ids=True, draw_nodes='connections', label_nodes=True)`
Here is what I get on my screen.
[image]

Thanks for working on this package.  I find it very nice for making Circuits exams.

Rob

frohro, 2019-12-08:
I have a question about the above circuit.  If I change the draw command to show primary nodes, it gives me none.  I assumed primary nodes were those without subscripts.  Maybe I'm wrong there.

`cct2.draw(label_ids=True, draw_nodes='connections', label_nodes='primary')`

Is primary the way to get nodes without subscripts printed?

Thanks,
Rob

mph-, 2019-12-09:
Hello Rob

Unfortunately this is an aliasing artefact during pdf to png conversion with pdf2ppm.  If you save to a .pdf file you should see that the lines are present.  Lcapy used to use the imagemagick convert program.  This did not have aliasing artefacts but disallowed pdf to png conversion unless you edited a system file.  

I've switched Lcapy to try convert first and then fall back on pdftoppm.  I would welcome any suggestions to improve this.

I've also fixed the problem with labelling primary nodes.  Originally, the purpose of the secondary nodes was for drawing and I didn't want them cluttering the schematic.  Foolishly, I originally chose to distinguish secondary nodes from primary nodes using an underscore in the name, not considering that some people wanted to use nodes with subscripts and superscripts.  With the new naming system,  secondary nodes either start with an underscore or start with a number followed by an underscore.

Michael.

closed 2019-12-09 by mph-