mph-/lcapy #29: Schematics showing very small in Jupyter notebook

jrbrodie77, 2020-01-24:
Love this tool!  When I create a schematic with basic operations, like (R1 + R2) | C1, in a jupy notebook, the schematic renders at a decent size.  When i use a netlist passed to Circuit, I get a very tiny image in the notebook.

See attached PNG for minimal example.

Also, are you guys taking pull requests? I've run into the odd error that I could likely fix and submit, if you'll have them.  (for instance:  I was missing one of the conversion utilities, the error msg didn't indicate what was wrong, I could add code to detect this.)

[image]

mph-, 2020-01-24:
Hello, I've pushed a patch for this.  It's all rather unsatisfactory how jupyter displays png files.  I have added some ad-hoc scaling and would appreciate any ideas how to improve this.  Ideally I would like to use svg but there is a jupyter bug in how they are displayed.

There is no fundamental difference between how networks and netlists are drawn except that the former adds short wires between components in series.

Any pull requests will be gratefully accepted; there's only me behind the scenes...

closed 2020-01-24 by mph-

mph-, 2020-01-26:
Update: I've removed the adhoc-ery with a dpi argument to the draw command.  This can also be specified in the netlist, for example, ; dpi=100