mph-/lcapy #30: Hard to understand error if GS not installed

mwThinks, 2020-03-17:
If you use Circuit.draw() without having ghostscript installed you get the following error

```bash
C:\ProgramData\Anaconda3\lib\site-packages\lcapy\schematic.py in png_image_size(filename)
     81 
     82     if (header[:8] != b'\211PNG\r\n\032\n' and (header[12:16] != b'IHDR')):
---> 83         raise Exception('%s not a png image' % filename)
     84     w, h = struct.unpack('>LL', header[16:24])
     85     width = int(w)

Exception: C:\Users\tiqi-F24\AppData\Local\Temp\tmp7nqsmvo8.png not a png image
```

which took me a long time to debug. Putting some simple check in the convert_pdf_png function in system.py if ghost script is actually installed might be a good idea.

I am using lcapy version 0.46.1 on Windows 10.

Thanks, all the best,
Martin

mph-, 2020-04-03:
Hello Martin

Yes, that's an excellent suggestion.  Indeed, it was on my list of things to do.  I was planning on probing which tools were available and caching the result.  

I know nothing about Windows bu would like to document what to install to get Lcapy with Windows.  Are you able to tell me how you installed all the tools needed to make it work?

Regards, Michael

mwThinks, 2020-04-03:
Hi Michael, 
thanks for getting back on this. If I remember correctly I just did a standard install of ghostscript ([https://www.ghostscript.com/download/gsdnld.html]).

I guess the problem is this:

```
def convert_pdf_png(pdf_filename, png_filename, dpi=300):
    try:
        convert_pdf_png_ghostscript(pdf_filename, png_filename, dpi)
    except:
        try:
            convert_pdf_png_convert(pdf_filename, png_filename, dpi)
        except:
            convert_pdf_png_pdftoppm(pdf_filename, png_filename, dpi)
```
Looks like you are just excepting all the errors which the calls to possibly non existing programs would raise. That works only if there is at least one of the programs installed.

Thanks for the great work on the package,
Martin

mph-, 2020-04-03:
Sorry for the slow response; the past couple of weeks have been rather unusual...
I've pushed a change to be more informative when running external programs.   Thanks for your help.

Cheers, Michael

closed 2020-04-03 by mph-

mwThinks, 2020-04-04:
Nice! Thanks for that!
Best, Martin