mph-/lcapy #33: sExpr in lcapy: cannot be used with sympy solvers?

xinyin8, 2020-06-24:
I am trying the lcapy package with sympy solvers and encountered the following errors, 

```
In [27]: from lcapy import s
In [28]: from sympy import simplify, collect, solve

In [29]: simplify(s+2*s+1)
Out[29]: 3⋅s + 1

In [30]: collect(s+2*s+1,s)
Out[30]: 3⋅s + 1

In [31]: solve(s+2*s+1,s)
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-31-266b3ed08da2> in <module>
----> 1 solve(s+2*s+1,s)

...\site-packages\sympy\solvers\solvers.py in solve(f, *symbols, **flags)
    968             continue  # neither re(x) nor im(x) will appear
    969         # if re(s) or im(s) appear, the auxiliary equation must be present
--> 970         if any(fi.has(re(s), im(s)) for fi in f):
    971             irf.append((s, re(s) + S.ImaginaryUnit*im(s)))
    972     if irf:

...\site-packages\sympy\solvers\solvers.py in <genexpr>(.0)
    968             continue  # neither re(x) nor im(x) will appear
    969         # if re(s) or im(s) appear, the auxiliary equation must be present
--> 970         if any(fi.has(re(s), im(s)) for fi in f):
    971             irf.append((s, re(s) + S.ImaginaryUnit*im(s)))
    972     if irf:

TypeError: has() takes 2 positional arguments but 3 were given
```

It seems that lcapy.sExpr e.g. 's' can work with sympy.simplify() and sympy.collect(), but it failed with sympy.solve(). Could it be used in this way or there is an alternative way? Thanks. 

Kind regards,
Xin

mph-, 2020-06-24:
Thanks for letting me know Xin.  Unfortunately, every time SymPy changes its API, these niggles turn up that I cannot test for.

Michael

closed 2020-06-24 by mph-

xinyin8, 2020-06-25:
Hi Michael, 

Thank you for the quick response. With the new commit, the error is gone now but the result is still not what I expected. 

> In [1]: from lcapy import s
> In [2]: from sympy import simplify, collect, solve
> In [3]: solve(s+2*s+1,s)
> Out[3]: []

I would expect it returns  [-1/3] as the following, 

> In [9]: x = Symbol('x')
> In [10]: solve(x+2*x+1,x)
> Out[10]: [-1/3]

Could you repeat what I had for solve(s+2*s+1,s)? Thanks. 

Kind regards,
Xin

mph-, 2020-06-25:
Oh dear, I'm sorry. It appears that I forgot to push my local changes.   Could you try again please?  If it still does not work, please send me the output from running show_version().

Regards, Michael.

xinyin8, 2020-06-29:
Hi Michael, 

It works. Thanks a lot!

Kind regards, Xin