mph-/lcapy #34: How to generically change the value of a simple element!

Dakantz, 2020-09-28:
Hi!

How would one simply change the the value of e.g. a resistor dynamically?

Regards,
Benedikt

mph-, 2020-09-29:
It depends on what you are trying to do.   You can use the subs method to replace its symbolic value with a numerical value.  However, if you wish to solve a circuit with a resistance that changes with time, then this cannot be currently done.  If the circuit was purely resistive the analysis is straightforward and Lcapy could compute the solution.  However, Lcapy expects resistors to have non-time varying resistances.  A work around would be to use an impedance given by the Laplace transform of your time dependent resistance.

Regards, Michael.

dyc3, 2021-05-16:
I think hes asking how to set a value (ohms, amps, volts, etc) of a component after it has already been placed into a circuit.

eg, something like
```python
cct = Circuit("""
Vs 1 0; down 
R1 1 2; right 
R2 2 0_1; down 
W 0 0_1; right
""")
cct.R1.value = 20
```

Is there a way to do that?

mph-, 2021-05-17:
The components are immutable but you can create a new circuit with the `subs()` method:

`newcct = cct.subs({'R1': 10})
`
This works for VCVS and CCCS.  

You can access the resistance of a resistor with the `.R` attribute:
`cct.R1.R`

I quite like your suggestion of a `.value` attribute but I will have to ponder all the unexpected consequences of making this settable.  Moreover, some components have multiple values; for example, a capacitor has capacitance and initial voltage attributes.  Do you have any suggestions of how to deal with them?

dyc3, 2021-05-18:
No, not really. After getting more familiar with the library, honestly I think that `.value` will ultimately be more confusing than how `subs` works now.

mph-, 2021-05-18:
Okay, thanks.

closed 2021-05-18 by mph-