mph-/lcapy #35: Can't apply initial conditions in circuit simulation

SanchoPansa47, 2020-11-02:
Hello,

Here is a circuit that fails to simulate:

from lcapy import Circuit
import numpy as np
import matplotlib.pyplot as plt

```
cct = Circuit('''
... Vin 1 0 {sin(130e6*2*3.14159*t)}
... Rin 1 2 50
... Cs  2 4 2.34e-12 0
... La  2 3 0.32e-6 0
... Rs  3 4 0.532
... Rl  4 0 50''')

##cct1 = cct.subs({'Rin': 50, 'Cs':2.34e-12, 'La': 0.32e-6, 'Rs': 0.532, 'Rl': 50})
##H = cct.transfer(1,0,4,0)
##print(H)

t = np.linspace(0, 2e-7, 1000)
Vinp = cct.Vin.v.evaluate(t)
Vout = cct.Rl.v.evaluate(t)

plt.plot(t, Vinp, 'g', linewidth=2)
plt.plot(t, Vout, 'r', linewidth=2)
plt.grid(b=True, which='major', color='#666666', linestyle='-')
plt.minorticks_on()
plt.grid(b=True, which='minor', color='#999999', linestyle='-', alpha=0.2)
plt.show()
```

mph-, 2020-11-02:
Thanks for this.  It is due to a bug in SymPy.  I've pushed a workaround to Lcapy.

Michael.

SanchoPansa47, 2020-11-03:
I've tried once more.
The problem persisits.

mph-, 2020-11-03:
That's weird, I get 
[image]

What does the following give for you?

`from lcapy import show_version
show_version()`

SanchoPansa47, 2020-11-03:
>>> from lcapy import show_version
>>> show_version()
Python: 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0]
SymPy: 1.6.2
NumPy: 1.19.2
Matplotlib: 3.3.2
SciPy: 1.5.3
Lcapy: 0.67.0
>>>

mph-, 2020-11-03:
Ah, you've got 0.67; you need to upgrade to 0.69

SanchoPansa47, 2020-11-03:
How to upgrade ?

SanchoPansa47, 2020-11-03:
Works !
Thanks

SanchoPansa47, 2020-11-03:
Another issue. How to pass parameters and in the same time specify initial conditions.
For example with the syntax below the simulation is erroneous.
```
from lcapy import Circuit
import numpy as np
import matplotlib.pyplot as plt

##cct = Circuit('''
##... Vin 1 0 {sin(130e6*2*3.14159*t)}
##... Rin 1 2 50
##... Cs  2 4 2.34e-12 0
##... La  2 3 0.32e-6 0
##... Rs  3 4 0.532
##... Rl  4 0 50''')

cct = Circuit('''
... Vin 1 0 {sin(130e6*2*3.14159*t)}
... Rin 1 2
... Cs  2 4 0
... La  2 3 0
... Rs  3 4
... Rl  4 0''')

cct1 = cct.subs({'Rin': 50, 'Cs':2.34e-12, 'La': 0.32e-6, 'Rs': 0.532, 'Rl': 50})
##H = cct.transfer(1,0,4,0)
##print(H)

t = np.linspace(0, 2e-7, 1000)
Vinp = cct1.Vin.v.evaluate(t)
Vout = cct1.Rl.v.evaluate(t)

plt.plot(t, Vinp, 'g', linewidth=2)
plt.plot(t, Vout, 'r', linewidth=2)
plt.grid(b=True, which='major', color='#666666', linestyle='-')
plt.minorticks_on()
plt.grid(b=True, which='minor', color='#999999', linestyle='-', alpha=0.2)
plt.show()
```

closed 2020-11-06 by mph-