mph-/lcapy #36: Arguments of draw() method not working as expected

mniehus, 2020-11-03:
hi. according to http://lcapy.elec.canterbury.ac.nz/modules.html#lcapy.netlist.NetlistMixin.draw
we should see some effects on the graphics with the following jupyter code: 

`%matplotlib inline
from lcapy import R, C, Par
#http://lcapy.elec.canterbury.ac.nz/modules.html#lcapy.netlist.NetlistMixin.draw
#n = C('C1') | (R('R1') + (C('C2') | (R('R2') + (C('C3') | (R('R3') + C('C4'))))))
n = R(1e3)+Par(R(1e3),R(10e3),R(.22e3))

#n.draw()
n.draw(form='horizontal',label_ids=True,label_nodes=False)
# seems no difference between horizontal, vertical, ladder? how get both terminals on left side?
#n.draw(form='vertical')  
#n.draw(form='ladder')
# Possible to only label the "nets" instead of all terminals ? 'pins' argument ?
n.draw(form='horizontal',label_ids=True,label_nodes=True)
# n.draw(form='horizontal',label_ids=True,label_nodes='pins')`

Should not "vertical" show up different than "horizontal"? 
And should not the "pins" annotate the "nets"?

Find attached a screenshot.
[image]

thanks and all the best to the antipode :)

mph-, 2020-11-03:
Hello, I think you have an older version of Lcapy.

What does the following give? 

```
from lcapy import show_version
show_version()
```

mniehus, 2020-11-03:
it gives:

Python: 3.7.7 (default, Mar 26 2020, 15:48:22)
[GCC 7.3.0]
SymPy: 1.6.2
NumPy: 1.19.1
Matplotlib: 3.3.1
SciPy: 1.5.2
Lcapy: 0.67.0

Links:
------
[1] https://github.com/mph-/lcapy/issues/36#issuecomment-720961453
[2] 
https://github.com/

mph-, 2020-11-03:
Ah, this was introduced in version 0.68.  You will need to update.

Michael.

closed 2020-11-03 by mph-