mph-/lcapy #37: Another issue. How to pass parameters and in the same time specify initial conditions.

mph-, 2020-11-03:
Another issue. How to pass parameters and in the same time specify initial conditions.
For example with the syntax below the simulation is erroneous.
```
from lcapy import Circuit
import numpy as np
import matplotlib.pyplot as plt

##cct = Circuit('''
##... Vin 1 0 {sin(130e6*2*3.14159*t)}
##... Rin 1 2 50
##... Cs  2 4 2.34e-12 0
##... La  2 3 0.32e-6 0
##... Rs  3 4 0.532
##... Rl  4 0 50''')

cct = Circuit('''
... Vin 1 0 {sin(130e6*2*3.14159*t)}
... Rin 1 2
... Cs  2 4 0
... La  2 3 0
... Rs  3 4
... Rl  4 0''')

cct1 = cct.subs({'Rin': 50, 'Cs':2.34e-12, 'La': 0.32e-6, 'Rs': 0.532, 'Rl': 50})
##H = cct.transfer(1,0,4,0)
##print(H)

t = np.linspace(0, 2e-7, 1000)
Vinp = cct1.Vin.v.evaluate(t)
Vout = cct1.Rl.v.evaluate(t)

plt.plot(t, Vinp, 'g', linewidth=2)
plt.plot(t, Vout, 'r', linewidth=2)
plt.grid(b=True, which='major', color='#666666', linestyle='-')
plt.minorticks_on()
plt.grid(b=True, which='minor', color='#999999', linestyle='-', alpha=0.2)
plt.show()
```

_Originally posted by @SanchoPansa47 in https://github.com/mph-/lcapy/issues/35#issuecomment-721104952_

mph-, 2020-11-03:
This is notation problem.   The first netlist specifies transient analysis; the second specifies phasor analysis.  

In the first netlist you specify the initial current for the inductor and the initial voltage for a capacitor, thus, Lcapy solves the circuit as an initial value problem:

    `>>> cct = Circuit("""
     Vin 1 0 {sin(130e6*2*3.14159*t)}
     Rin 1 2 50
     Cs  2 4 2.34e-12 0
     La  2 3 0.32e-6 0
     Rs  3 4 0.532
     Rl  4 0 50""")

     >>> cct.describe()
    This has initial conditions so is an initial value problem solved in the s-domain using Laplace transforms.

    >>> cct.analysis                                                            
    {'zeroic': True,
    'has_ic': True,
    'ivp': True,
    'has_dc': False,
    'has_ac': False,
    'has_s': False,
    'has_transient': True,
    'reactances': ['Cs', 'La'],
    'dependent_sources': [],
    'independent_sources': ['Vin'],
    'control_sources': [],
    'ac': False,
    'dc': False,
    'causal': False,
    'time_domain': False}`

For the second netlist, no initial conditions are specified:

    `>>> cct = Circuit('''
    Vin 1 0 {sin(130e6*2*3.14159*t)}
    Rin 1 2
    Cs  2 4 0
    La  2 3 0
    Rs  3 4
    Rl  4 0''')

    >>> cct.describe()                                                         
    Phasor analysis is used for source Vin.

    {'zeroic': True,
    'has_ic': False,
    'ivp': False,
    'has_dc': False,
    'has_ac': False,
    'has_s': False,
    'has_transient': True,
    'reactances': ['Cs', 'La'],
    'dependent_sources': [],
    'independent_sources': ['Vin'],
    'control_sources': [],
    'ac': False,
    'dc': False,
    'causal': False,
    'time_domain': False}`

I would appreciate any suggestions of how to clarify or make it more intuitive.

Michael.

SanchoPansa47, 2020-11-04:
Didn't understand. Is there a way to describe a circuit using parameters (i.e. Cs, La, etc.) and in the same time apply initial conditions for C and L. Then substitute the parameters with real values using **subs**. What would be the syntax for such scenario ?

mph-, 2020-11-04:
`>>> cct = Circuit("""
     Vin 1 0 {sin(130e6*2*3.14159*t)}
     Rin 1 2 50
     Cs  2 4 2.34e-12 v0
     La  2 3 0.32e-6 i0
     Rs  3 4 0.532
     Rl  4 0 50""")

     >>> cct2 = cct.subs({'i0':1, 'v0':2})
     >>> cct2.La.V(t)
     # or
     >>> cct.La.V(t).subs({'i0':1, 'v0':2})`

closed 2020-11-04 by mph-

SanchoPansa47, 2020-11-05:
In your example you specified components values as constants and initial conditions as symbols. I search for inverse solution: component values - symbols, initial values - constants. Possible ?

mph-, 2020-11-06:
Sorry, I don't fully understand your question.

You can define a capacitor with capacitance C and initial voltage v0 as
   `C1 1 2 C v0`

If you know the initial value, say 0.1 V, you can either substitute the symbolic value or define the capacitor as

   `C1 1 2 C 0.1`

MIchael.

SanchoPansa47, 2020-11-09:
Works. Thanks !