mph-/lcapy #39: Convergence in netlist analysis of 3-loop topology

sikersten, 2020-11-11:
I'd like to analyse a netlist with a 3-loop topology, e.g.
`from lcapy import Circuit, s

cct= Circuit("""
V1  1   0;
R1  1   2;
R2  2   0;
R3  1   3;
R4  3   0;
R5  2   3;
""")
H = cct.R1.V(s)`

Unfortunately, it seems like the computation does not converge. What could be the reason and how to deal with it?

mph-, 2020-11-11:
Hello, what answer do you get?

I get

 `R1*(R2*R3 + R3*R4 + R3*R5 + R4*R5)*V1(s)/(R1*R2*R3 + R1*R2*R4 + R1*R3*R4 + R1*R3*R5 + R1*R4*R5 + R2*R3*R4 + R2*R3*R5 + R2*R4*R5)`

Note, if you know V1 is a dc source you can define at as `V1 1 0 dc`

sikersten, 2020-11-12:
Thanks for your reply! The computation did not converge (I interrupted it after an hour or so), so I did not get a solution. I just set up a new environment with the minimum requirements, it works fine now (solves within some seconds). So it seems like the environment was somehow faulty, though I could not detect the actual problem.

closed 2020-11-12 by sikersten

mph-, 2020-11-12:
Perhaps you were using sympy-1.6.1 ?  This had a serious time regression for matrix inverses, see
https://github.com/sympy/sympy/issues/19920