mph-/lcapy #40: Mirroring transistors does not work

fakuivan, 2020-11-13:
Mirroring transistors on ``lcapy (0.63), python 3.8.0`` causes them to swap the inputs but the symbol stays the same:

[image]

This is on a blank jupyter notebook with only that code and ``import lcapy`` on the top.

mph-, 2020-11-14:
Thanks for letting me know.  I've pushed a patch for this.  You will need to use the invert attribute to flip horizontally.

fakuivan, 2020-11-15:
Thanks for the reply! I tried using the ``invert`` attribute but it errored out telling me the component cannot be flipped.

mph-, 2020-11-15:
Hello, did you install from git?  If so, what does the following give? 
 
  `from lcapy import show_version;   show_version()`

I'm going to generate a new pip release later today...

fakuivan, 2020-11-16:
I've installed lcapy from pip, running from wsl1. Here's the version info:

```text
fakui@DESKTOP-TN8UDAO:~$ python3.8 -c 'from lcapy import show_version; show_version()'
Python: 3.8.0 (default, Oct 28 2019, 16:14:01)
[GCC 8.3.0]
SymPy: 1.6.1
NumPy: 1.19.1
Matplotlib: 3.3.0
SciPy: 1.5.2
Lcapy: 0.63
```

mph-, 2020-11-17:
I've generated a new release (V0.70) you can download from pip.  See also the updated docs.

closed 2020-11-17 by mph-

fakuivan, 2020-11-17:
I've installed 0.70 from pip and I can confirm the error has been solved

[image]

[image]

Thank you for the help!