mph-/lcapy #46: Invalid Latex code for state space equations ("Double Subscript Problem")

tqml, 2021-01-01:
Lcapy v0.74 outputs invalid Latex code for state space equatiosn when components use  subscripts.

## Steps to reproduce

Take an example circuit, e.g.:

```python
a = Circuit("""
Vstep 1 0 {v(t)};  down
R1 1 2; right
L 2 3; right=1.5, i={i_L}
R2 3 0_3; down=1.5, i={i_{R2}}, v={v_{R2}}
W 0 0_3; right
W 3 3_a; right
C 3_a 0_4; down, i={i_C}, v={v_C}
W 0_3 0_4; right""")
ss = a.ss
ss.state_equations()
```
Change the name of `C` to `Co`or `C_o`. Then print the state space equations. The result is then not displayed properly because the generated code is not valid Latex code ('double subscript' problem)

[image]

Replacing `v_C_o` with `v_{C_o}` fixes the Latex error. I have not found any indication in the documentation that this is actual expected behaviour so I think it is a bug. Also other components with subscripts use proper braces and only state variables are affected.

##  Versions

Python 3.7, lcapy  v0.74

mph-, 2021-01-08:
Thanks for letting me know.  The double subscript problem should really be be fixed in SymPy but I have pushed a work-around to master. 

As a caveat, I have also pushed many other changes related to a restructuring of the internal expression classes.  These changes pass all the current tests but I am sure that I may have broken an existing use case.  I'm holding off making a new release until I am more confident that I won't get a flood of issues.

closed 2021-01-11 by mph-

tqml, 2021-01-13:
Yes, I'm very sorry after posting the issue I saw that there was already a discussion on this topic. 
Anyway thank you for the quick fix and the amazing library!