mph-/lcapy #47: Cannot assign nodes which are not even added to the circuit.

lucatatas, 2021-02-03:
Hi,
I am getting the following error:

This error is listing nodes I have not added to my circuit. 
Here is an a list of which nodes I have added:
[image]
[image]

I don't really know if this is just a simple bug of if I am using the library incorrectly.

Thanks in advance

mph-, 2021-02-04:
Hello Luca

This is probably a bug and unfortunately this part of the code can be really tricky to debug.  Are you able to generate a simpler test case?

Michael

mph-, 2021-02-09:
Could you please add the failing code but not as an image?

Thanks, Michael

lucatatas, 2021-02-09:
Hi,I have just been reading this mail. I have found a workaround for the problem, but I can’t really share my code, because my code generates a random circuit so the problem is not really reproducible. Thanks anyway. RegardsLuca Tatas Von: mph-Gesendet: Dienstag, 9. Februar 2021 21:49An: mph-/lcapyCc: lucatatas; AuthorBetreff: Re: [mph-/lcapy] Cannot assign nodes which are not even added to the circuit. (#47) Could you please add the failing code but not as an image?Thanks, Michael

mph-, 2021-02-22:
The master branch should no longer fail for this although the schematic may not have the desired symmetry.  This is a gnarly problem to solve since heuristics are required.  Instead, Lcapy now prints a message suggesting that a constraint is added between pairs of nodes (such as an open-circuit component) to enforce symmetry.

Michael.

closed 2021-02-22 by mph-