mph-/lcapy #48: No module named 'sympy.physics.units.systems.si'

drscotthawley, 2021-02-06:
Hi, I just learned about lcapy, but the installation instructions don’t work for me.  Perhaps sympy has changed.  What (older) version of sympy does one need?  I couldn’t find any info on this in the documentation, and there doesn’t seem to be a `requirements.txt` file. 

```
$ pip installed lcapy 
Requirement already satisfied: sympy in /usr/local/lib/python3.6/dist-packages (1.1.1)
Requirement already satisfied: lcapy in /usr/local/lib/python3.6/dist-packages (0.77)
Requirement already satisfied: mpmath>=0.19 in /usr/local/lib/python3.6/dist-packages (from sympy) (1.1.0)
Requirement already satisfied: numpy in /usr/local/lib/python3.6/dist-packages (from lcapy) (1.19.5)
Requirement already satisfied: matplotlib in /usr/local/lib/python3.6/dist-packages (from lcapy) (3.2.2)
Requirement already satisfied: wheel in /usr/local/lib/python3.6/dist-packages (from lcapy) (0.36.2)
Requirement already satisfied: scipy in /usr/local/lib/python3.6/dist-packages (from lcapy) (1.4.1)
Requirement already satisfied: setuptools in /usr/local/lib/python3.6/dist-packages (from lcapy) (53.0.0)
Requirement already satisfied: IPython in /usr/local/lib/python3.6/dist-packages (from lcapy) (5.5.0)
Requirement already satisfied: networkx in /usr/local/lib/python3.6/dist-packages (from lcapy) (2.5)
Requirement already satisfied: python-dateutil>=2.1 in /usr/local/lib/python3.6/dist-packages (from matplotlib->lcapy) (2.8.1)
Requirement already satisfied: cycler>=0.10 in /usr/local/lib/python3.6/dist-packages (from matplotlib->lcapy) (0.10.0)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in /usr/local/lib/python3.6/dist-packages (from matplotlib->lcapy) (2.4.7)
Requirement already satisfied: kiwisolver>=1.0.1 in /usr/local/lib/python3.6/dist-packages (from matplotlib->lcapy) (1.3.1)
Requirement already satisfied: decorator in /usr/local/lib/python3.6/dist-packages (from IPython->lcapy) (4.4.2)
Requirement already satisfied: pygments in /usr/local/lib/python3.6/dist-packages (from IPython->lcapy) (2.6.1)
Requirement already satisfied: prompt-toolkit<2.0.0,>=1.0.4 in /usr/local/lib/python3.6/dist-packages (from IPython->lcapy) (1.0.18)
Requirement already satisfied: pexpect; sys_platform != "win32" in /usr/local/lib/python3.6/dist-packages (from IPython->lcapy) (4.8.0)
Requirement already satisfied: pickleshare in /usr/local/lib/python3.6/dist-packages (from IPython->lcapy) (0.7.5)
Requirement already satisfied: simplegeneric>0.8 in /usr/local/lib/python3.6/dist-packages (from IPython->lcapy) (0.8.1)
Requirement already satisfied: traitlets>=4.2 in /usr/local/lib/python3.6/dist-packages (from IPython->lcapy) (4.3.3)
Requirement already satisfied: six>=1.5 in /usr/local/lib/python3.6/dist-packages (from python-dateutil>=2.1->matplotlib->lcapy) (1.15.0)
Requirement already satisfied: wcwidth in /usr/local/lib/python3.6/dist-packages (from prompt-toolkit<2.0.0,>=1.0.4->IPython->lcapy) (0.2.5)
Requirement already satisfied: ptyprocess>=0.5 in /usr/local/lib/python3.6/dist-packages (from pexpect; sys_platform != "win32"->IPython->lcapy) (0.7.0)
Requirement already satisfied: ipython-genutils in /usr/local/lib/python3.6/dist-packages (from traitlets>=4.2->IPython->lcapy) (0.2.0)
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-17-3ab9595a4556> in <module>()
      1 # trying circuit-drawing for a second...
      2 get_ipython().system('pip install sympy lcapy')
----> 3 from lcapy import Circuit
      4 
      5 cct = Circuit()

3 frames
/usr/local/lib/python3.6/dist-packages/lcapy/units.py in <module>()
      8 
      9 import sympy.physics.units as u
---> 10 from sympy.physics.units.systems.si import dimsys_SI
     11 from sympy.physics.units.systems import SI
     12 from sympy.physics.units import UnitSystem

ModuleNotFoundError: No module named 'sympy.physics.units.systems.si'
```

drscotthawley, 2021-02-06:
Update: Downgraded all the way back to `sympy==0.5.14`, still get the exact same error.

Not sure how this problem not covered in the troubleshooting docs.  Seems like this would happen to everyone trying to use this package.

mph-, 2021-02-06:
I'm not sure what is happening with your setup since the tests work fine with ubuntu-16.04, ubuntu-18.04, ubuntu-20.04, macos-10.15, and windows-2019 using Python versions  3.6, 3.7, 3.8, and 3.9.

What happens if you try:
`>>>  from sympy.physics.units.systems.si import dimsys_SI`

Perhaps also try:
 `>>> import sympy.physics.units.systems.si`
 `>>> sympy.physics.units.systems.si.__file__`

drscotthawley, 2021-02-07:
This is on Ubuntu 18.4 with Python 3.6.9, which is what Google Colab runs. There error says that sympy.physics.units.systems.si does not exist, so both of those commands you suggest produce the same error as before. 

What version of sympy is it expecting? I notice from the pip install that it’s version 1.11.1.

Here is a shareable link to a minimal Colab notebook: https://colab.research.google.com/drive/1Gub_Kb676tAzAaYRLnYbZOBxyvwLDTQe?usp=sharing

drscotthawley, 2021-02-07:
Woops, pressed a wrong button, did not mean to close this.  Reopening.

drscotthawley, 2021-02-07:
I see that 'mks' exists, just 'si':

```
>>> import sympy.physics.units.systems as systems
>>> dir(systems)
['MKS',
 'MKSA',
 '__builtins__',
 '__cached__',
 '__doc__',
 '__file__',
 '__loader__',
 '__name__',
 '__package__',
 '__path__',
 '__spec__',
 '_mks_dim',
 '_mksa_dim',
 '_natural_dim',
 'mks',
 'mksa',
 'natural']
```

mph-, 2021-02-07:
For some reason sympy-1.1.1 was being installed by default.  I've set the minimum version to sympy-1.7.1 and have released lcapy-0.80.   Hopefully, this will do the trick.

drscotthawley, 2021-02-07:
Thanks! Problem no longer occurs in Colab. Closing.

closed 2021-02-07 by drscotthawley