mph-/lcapy #49: Expression error when Current sources used in circuit

abnaidu001, 2021-03-17:
**Code:**
from lcapy import Circuit,t
cct = Circuit("""
Ia 1 0;
RBLK 1 2;
CBLK 2 0;
""")
ss=cct.ss
**Error:**

  File "C:\Users\a0230109\AppData\Local\Continuum\anaconda3\lib\site-packages\sympy\parsing\sympy_parser.py", line 1008, in parse_expr
    return eval_expr(code, local_dict, global_dict)

  File "C:\Users\a0230109\AppData\Local\Continuum\anaconda3\lib\site-packages\sympy\parsing\sympy_parser.py", line 903, in eval_expr
    code, global_dict, local_dict)  # take local objects in preference

  File "<string>", line 1
    <Symbol ('built' ,positive=True )-Symbol ('in' ,positive=True )Symbol ('method' ,positive=True )Symbol ('lower' ,positive=True )Symbol ('of' ,positive=True )Symbol ('str' ,positive=True )Symbol ('object' ,positive=True )Symbol ('at' ,positive=True )Integer (0x000002A7A9ECEDF8 )>(t )
    ^
SyntaxError: invalid syntax

-------------------------------------------------------------

Facing this error only when I use Current sources, No issues when voltage sources used. All package versions seem to be up-to-date.

mph-, 2021-03-18:
Thanks for this.  I've pushed a simple fix for this.

abnaidu001, 2021-03-18:
Thanks. This fixed it.

closed 2021-03-18 by abnaidu001