mph-/lcapy #50: Incorrect state space equations with current source input

abnaidu001, 2021-03-18:
**Code:**

from lcapy import Circuit,t

cct = Circuit("""
I1 1 0;
CBLK 1 0;
""")
ss=cct.ss
----------------------------------------------------
**State space equations:**

ss.state_equations()
Out[28]: 
⎡d            ⎤                          
⎢──(v_CBLK(t))⎥ = [0]⋅[v_CBLK(t)] + []⋅[]
⎣dt           ⎦                          

ss.output_equations()
Out[29]: 
[v₁(t)] = [1]⋅[v_CBLK(t)] + []⋅[]

-------------------------------------------------
I don't see Input vector to be current source here. Also ss.B vector is empty. Please help.

mph-, 2021-03-18:
I need some help with this since I seem to get the correct answers:

```
In [28]: ss.A
Out[28]: [0]
In [29]: ss.B
Out[29]: 
⎡  1  ⎤
⎢─────⎥
⎣C_BLK⎦
In [30]: ss.C
Out[30]: [1]
In [31]: ss.D
Out[31]: [0]
```

Can you please send me what you get for:

```
 >>> from lcapy import show_version
 >>> show_version()
```

abnaidu001, 2021-03-19:
Hi,
Thanks.
Even I seem to be getting correct answers when I ran the same case after the restart of Console.
I don't know  what caused it to report incorrect equations earlier.  

ss.state_equations()
Out[6]: 
⎡d                     ⎤                                ⎡  1  ⎤        
⎢──(v_CBLK(t))⎥ = [0]⋅ [v_CBLK(t)] + ⎢─────⎥⋅[i₁(t)]
⎣dt                   ⎦                                 ⎣C_BLK⎦        

ss.output_equations()
Out[7]: 
[v₁(t)] = [1]⋅[v_CBLK(t)] + [0]⋅[i₁(t)]

ss.B
Out[8]: 
⎡  1  ⎤
⎢─────⎥
⎣C_BLK⎦

from lcapy import show_version

show_version()
Python: 3.7.3 
SymPy: 1.7.1
NumPy: 1.16.4
Matplotlib: 3.1.0
SciPy: 1.2.1
Lcapy: 0.82

closed 2021-03-19 by abnaidu001