mph-/lcapy #51: Permission Error

Veoooooo, 2021-03-24:
Hi
I want to execute this simple example but I couldn't do it 

code:

from lcapy import R, C, L
cct2= (R(1e6) + L(2e-3)) | C(3e-6)
cct2.draw()

and I get this error:
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'D:\\OneDrive - King Fahd University of Petroleum & Minerals (KFUPM)\\Electric Python\\tmppg8rlavz.log'

[image]

mph-, 2021-03-24:
Hello

Which version of lcapy are you running?

```
from lcapy import show_version
show_version()
```

Veoooooo, 2021-03-24:
this is the version :
[image]

Veoooooo, 2021-03-24:
this is when I try to run it on C drive:

[image]

mph-, 2021-03-24:
Thanks for this.

I've pushed a patch to master; could you please check that it works for you?

Veoooooo, 2021-03-24:
another error raised: 

[image]

mph-, 2021-03-24:
Did this use to work for V0.82?

Veoooooo, 2021-03-24:
actually, I tried the new patch on another machine and it is working perfectly.
[image]
Thanks a lot! 
All the best, Alshehri.

Veoooooo, 2021-03-24:
> Did this use to work for V0.82?

no

Veoooooo, 2021-03-25:
> another error raised:
> 
> [image]

I have found that there is a problem with the path of the files that lcapy produce 
and it is specific for the one drive path, if I use it some where else the lcapy library is working fine

closed 2021-03-25 by Veoooooo

mph-, 2021-03-25:
Would you mind trying a patch I've pushed to master and then record the output from:

```
from lcapy import R
R(1).draw(debug=2)
```
The problem may be because I changed from saving temporary files in a temp directory and use the current working directory instead.  I know very little about Windows but I suspect that the current working directory may not be valid.

Veoooooo, 2021-03-26:
yeah,  you are right this is working now thanks for your effort and I appreciate it.
[image]