mph-/lcapy #53: Unable to put equality into standard form

dyc3, 2021-05-16:
Here's my code

```python
cct_before = Circuit("""
C 1 0; down
W 1 1_1; right
W 0 0_1; right
R4 1_1 0_1; down
L 1_1 2; right
W 0_1 0_2; right
R5 2 0_2; down
W 2 2_1; right
W 0_2 0_3; right
Is 2_1 0_3; down

;draw_nodes=connections, cpt_size=1
""")

cct_after = Circuit("""
V1 1 0 dc Vs; down
R1 1 2; right
W 0 0_1; right
R2 2 0_1; down
W 2 2_1; right
W 0_1 0_2; right
R3 2_1 0_2; down
W 2_1 2_2; right
W 0_2 0_3; right
C 2_2 0_3; down

;draw_nodes=connections, cpt_size=1
""")
cct_before.draw()
cct_after.draw()

t1 = expr('t1', positive=True)
cct = cct_after.initialize(cct_before, t1)

n = NodalAnalysis(cct)
result = n.nodal_equations()
exp = result[2]
sympy.pprint(exp.standard()) # fails
```

Stack trace:

```
~/anaconda3/envs/school/lib/python3.9/site-packages/lcapy/expr.py in standard(self)
   2146         if self._ratfun is None:
   2147             return self.copy()
-> 2148         return self.__class__(self._ratfun.standard(), **self.assumptions)
   2149 
   2150     def mixedfrac(self):

~/anaconda3/envs/school/lib/python3.9/site-packages/lcapy/ratfun.py in standard(self, split)
    582 
    583         try:
--> 584             Q, M, D, delay, undef = self.as_QMD()
    585         except ValueError:
    586             if not split:

~/anaconda3/envs/school/lib/python3.9/site-packages/lcapy/ratfun.py in as_QMD(self)
    695         expression = (Q + M / D) * exp(-delay * var) * undef"""
    696 
--> 697         N, D, delay, undef = self.as_ratfun_delay_undef()
    698 
    699         # Perform polynomial long division so expr = Q + M / D

~/anaconda3/envs/school/lib/python3.9/site-packages/lcapy/ratfun.py in as_ratfun_delay_undef(self)
    319         Note, delay only represents a delay when var is s."""
    320 
--> 321         return as_ratfun_delay_undef(self.expr, self.var)
    322 
    323     def as_const_undef_rest(self):

~/anaconda3/envs/school/lib/python3.9/site-packages/lcapy/ratfun.py in as_ratfun_delay_undef(expr, var)
    264     undef = sym.S.One
    265 
--> 266     if expr.is_rational_function(var):
    267         N, D = as_numer_denom(expr, var)
    268         return N, D, delay, undef

AttributeError: 'Equality' object has no attribute 'is_rational_function'
```

mph-, 2021-05-17:
Thanks for the testcase.  It exposed two bugs!  I have pushed a fixed.

dyc3, 2021-05-17:
Thanks. FYI, it appears that v0.89 and v0.90 have not been pushed to pypi, and releases aren't being tagged here on github.

mph-, 2021-05-17:
Hmm, I thought I pushed v0.89 to pypi.  I'm pushing v0.90 tomorrow once I've added more tests for some of the new discrete time stuff.

closed 2021-05-17 by mph-