mph-/lcapy #55: Documentation's website is down

javovelez, 2021-05-19:


psychemedia, 2021-05-20:
Still down... any chance of you also making it available via Github pages, even if not explicitly linked...?

psychemedia, 2021-05-20:
I've never used readthedocs. What happens if you just enable Github pages from the repo settings and point it to main branch docs folder?

There are probably Github Actions that might be able to do the work too.

mph-, 2021-05-21:
Unfortunately, the site has been taken offline until a security audit has been completed.  In the meantime I have been trying to migrate to readthedocs.org but have been having trouble trying to build the images.  Do you have any experience with this?  I may have to build the images myself and add them to the repository.

mph-, 2021-05-21:
After much frustration, I've loaded the docs at https://lcapy.readthedocs.io/en/latest/  

Tomorrow, I'll try to get readthedocs to generate the images...

javovelez, 2021-05-21:
Sorry, I'd love to help but I don't have any experience on the subject.

psychemedia, 2021-05-21:
What's the issue re: generating the docs? Presumably you are managing to get the code to run?

I am using Jupyter Book for [this site](https://opencomputinglab.github.io/SubjectMatterNotebooks/) and it seems to work okay (I run Jupyter Book locally then copy the site files over. On to do list is the automation, but the environment is rather complex and I need to properly dockerise it.

mph-, 2021-05-21:
I ended up having to include the images in the repo.  I could not find a way to run all the scripts on readthedocs and some take a long time to generate.

closed 2021-05-30 by mph-