mph-/lcapy #56: decorator package pinned at decorator==4.4.2

psychemedia, 2021-07-13:
Hi

In the setup, you pin decorator to a very specific, now outdated version: https://github.com/mph-/lcapy/blob/df41025841cadfedb04e4847442b8049f0809a2f/setup.py#L26

[Current version](https://pypi.org/project/decorator/) looks to be at `v5.0.9`.
 
Any chance of relaxing the current constraint?

mph-, 2021-07-13:
Hi can you suggest which version to use?  I have tried removing the constraint and making the constraint >= 4.4.2 but both those options give dependency conflicts with networkx.

psychemedia, 2021-07-13:
Hmmm.. so where does that depnedency arise? I note from a [`(very?) recent `networkx` release note](https://github.com/networkx/networkx/blob/5d57c095466c742c216f4143ac66abe01367728e/doc/release/release_2.6.rst) that *"NetworkX no longer depends on the library "decorator"."*?

mph-, 2021-07-14:
I'm not sure.   The good news is that after removing the requirement for the decorator library, everything still builds.

closed 2021-07-15 by mph-