mph-/lcapy #57: Warning messages although Circuit is drawing fine

lucatatas, 2021-08-04:
Hi, I have the following questino/issue: Sometims Lcapy prints some warning like the ones shown in the attached screenshot. Because these only seem to be warnings rather than actual errors, I don't really want them to be printed. So I'd like to know, where the print statement is located which prints these warning. Sadly I was not able to finde them on my own.
Thanks in advance!
[image]

mph-, 2021-08-09:
These are printed in lcapy/schemgraph.py; grep is your friend!

For your example, you are being warned about getting a wonky schematic.

I have pondered about using the python warnings subsystem so that specific warnings can be disabled but will need a rainy day to implement.

closed 2021-08-10 by lucatatas

lucatatas, 2021-08-10:
Hi, thank you for your answer, another question which came to my mind was the following: What are these nodes called "Wanon16" or "Wanon27" for example? I have not specifically added those so I was wondering why they are sometimes part of my circuit.

mph-, 2021-08-10:
Each wire without an explicit name is given an anonymous name.  For example, W  1 2.