mph-/lcapy #62: Calling .draw() on a twoport.Chain yields TypeError: netlist() got an unexpected keyword argument 'layout'

jbm9, 2021-11-06:
Hi,

lcapy is really neat, I was writing a much, much more amateurish thing when i thought to google if anyone had made something similar, and wow did they ever.  It's great!  Thank you!

I'm running into the following problem (in a jupyter notebook, forgive the formatting):

```python
%pylab inline
import lcapy
print(f'lcapy version: {lcapy.__version__}')
n = lcapy.Series(lcapy.R('R_1')).chain(lcapy.Shunt(lcapy.R('R_2')))
n.draw()
```
Output:
```
Populating the interactive namespace from numpy and matplotlib
lcapy version: 0.99

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-36-695450633f62> in <module>
      3 print(f'lcapy version: {lcapy.__version__}')
      4 n = lcapy.Series(lcapy.R('R_1')).chain(lcapy.Shunt(lcapy.R('R_2')))
----> 5 n.draw()

/usr/local/lib/python3.8/dist-packages/lcapy/network.py in draw(self, filename, layout, form, evalf, **kwargs)
    213             layout = form
    214 
--> 215         self.sch(layout=layout, evalf=evalf).draw(filename=filename, **kwargs)
    216 
    217     @property

/usr/local/lib/python3.8/dist-packages/lcapy/network.py in sch(self, layout, evalf)
    159         """Convert a Network object into a Schematic object."""
    160 
--> 161         netlist = self.netlist(layout=layout, evalf=evalf)
    162         sch = Schematic()
    163         for net in netlist.split('\n'):

TypeError: netlist() got an unexpected keyword argument 'layout'
```

I apologize if this is expected behavior: there's kind of a lot to `lcapy`, and I suspect I'm still well below the inflection point on this learning curve.

This smells more like a bug than me doing something wrong, but, as I like to say, my ignorance knows no bounds.

Thanks again for putting all of this together, it's quite interesting!

mph-, 2021-11-06:
Thanks for you kind words.

Yes, your sense of smell is good and indeed it is a bug.  I've pushed a fix to github but Lcapy now raises an exception since I have yet to figure out an algorithm to convert a two-port abstract syntax tree to a netlist required for drawing.  If you have any suggestions, please let me know.

jbm9, 2021-11-06:
Ah, that makes sense.  I was thinking of it as a composition of objects, instead of as an arbitrary netlist.  Thanks for resolving this one quickly!

closed 2021-11-06 by jbm9

mph-, 2021-11-07:
I have added support for drawing Shunt and Series twoports and combinations using Chain.  For example,

`Chain(Series(Z('Z')), Shunt(Y('Y'))).draw()`

It was easier than I anticipated, although the other twoports will be more tedious.

mph-, 2021-11-14:
FYI, I have overhauled the two-port support; it is mostly described in the docs.