mph-/lcapy #63: Add triode component

danieljfarrell, 2021-11-22:
I'm interested in adding a Triode component, would that be of interest to the project?

It is basically a MOSFET with a different circuit symbols.

Reading the documentation, 

> Transistors (BJT, JFET, and MOSFET) can be drawn but not analyzed.

A different topic, but is it conceivable to do small-signal analysis of active components like MOSFET or Triode using linear approximations? For example, the equivalent circuit of a MOSFET are voltage controlled current sources with various parasitic capacitances (similar to the equivalent circuit of the Triode).

mph-, 2021-11-24:
Yes, I would welcome your contribution.

I'm not sure of what syntax should be used since there is no precedence with Spice.

Perhaps something like

`VTname Nanode Ngrid Ncathode triode [parameters]`

To add this you need to add a line to grammar.py and then create classes in schemcpts.py and mancpts.py.

Small-signal analysis is definitely achievable.  Indeed the model could be linearised around the DC level and this can be determined by Lcapy.   How complicated a model are you considering?

danieljfarrell, 2021-11-24:
Thanks for the pointers on the inner workings of the project! I will look into this.

As for the small signal analysis, it would good to calculate the voltage gain of a triode in a circuit and get the frequency response. 

Main parameters are,

1. the transconductance, g_m
2. The internal anode resistance, r_a 
3. Parasitic capacitances (Miller capacitance, etc)

I guess this must already be possible in Lcapy using the equivalent circuit of the triode.

danieljfarrell, 2021-11-24:
I'm looking at the drawing code in schemcpts.py

I was considering subclassing FixedCpt

```python
class Triode(FixedCpt):
    """Triode"""
    
    can_mirror = True
    can_scale = True
    can_invert = True   
```

I noticed that circuitikz has a triode component, https://github.com/circuitikz/circuitikz/issues/221 is there a way of reusing it directly?

mph-, 2021-11-24:
I suggest using the Gyrator class as an example; just drop one of the pins.  

The fiddly bit is determining the pin positions.  I do this by trial and error.  You can use helplines (see docs).

danieljfarrell, 2021-12-11:
I have started this using the Gyrator class.

mph-, 2021-12-22:
How did you get on with this?  Did you need a hand?

danieljfarrell, 2021-12-22:
I cloned the project, and have it all running locally. I’m adding unit tests for the triode, which are all failing! 

I will tidy up what I have and do a pull request so you can at least see what I’ve been doing.

I think firstly, just having the ability to generate schematics with triode symbol would be a good step. Then can make it more complicated.

But yes! Some help would be appreciated ;)

danieljfarrell, 2022-01-30:
Just to share what I am doing, so you can have a look, I have made a pull request.

I was going to suggest I close this issue, because I am currently working on other things and not too sure I can return to it for a while. But will leave that up to you.

danieljfarrell, 2022-01-30:
Pull request is #65