mph-/lcapy #64: Loop analysis shows `l_aplace` symbol

danieljfarrell, 2021-12-11:
I am running the example from the docs in latest version of lcapy.

Am I doing something stupid?

```python
import lcapy
print(lcapy.__version__)  # '1.1'

cct = lcapy.Circuit("""
...V1 1 0; down
...R1 1 2; right
...L1 2 3; right
...R2 3 4; right
...L2 2 0_2; down
...C2 3 0_3; down
...R3 4 0_4; down
...W 0 0_2; right
...W 0_2 0_3; right
...W 0_3 0_4; right""")
cct.draw()
```
[image]

```python
cct.laplace().loop_analysis().matrix_equations(form="A y = b")
```
[image]

mph-, 2021-12-22:
I have pushed a fix for this.  Sorry for the delay; somehow I lost the email notification.

danieljfarrell, 2021-12-22:
Great! Should I test it and close this issue? Or have you already tested it?

mph-, 2021-12-22:
Yip, it seems to be fixed.

closed 2021-12-22 by mph-