mph-/lcapy #66: Autotransformer component

danieljfarrell, 2022-01-30:
I would be interested to add a step-down autotransformer.

[image]

This is a single winding transformer.

In terms of the input and output currents, the behaviour is identical to a conventional isolation transformer.

The only difference is the current flowing in the common part of the winding (labeled `I_T` in the diagram is the difference between the current flowing in the primary and secondary windings,

[image]

This could be implemented easily as coupled inductors in series using mutual inductance (K) parameter [similar to the one that appears in Spice](https://www.analog.com/en/technical-articles/ltspice-basic-steps-for-simulating-transformers.html). Maybe? But don't know if this "K statement" exists in Lcapy.

I am mainly interested in using Lcapy to investigate the losses in such a transformer by including resistance of the winding. The primary winding resistance is constant, the secondary winding resistance changes depending on the step-down ratio. The secondary winding is the common section and has reduced current (see equations), so will have reduced losses.

mph-, 2022-02-01:
It would be straightforward to add an ideal autotransformer to Lcapy but it would work at DC!  If a autotransformer component was added, what parameters do you think it needs?

Yes, mutual inductances can be modelled with the K statement.

danieljfarrell, 2022-02-01:
I've actually been using lcapy recently to generate some analytical expressions with transformers (simply step-up, down stuff). I used an AC source, it worked very well!

I guess, a black box model of a step-down autotransformer looks identical to this,

```python
c = Circuit("""
Vs 1 0 ac; down
Rs 1 2; right=1.5
TF1 3 0_3 2 0_2 1/n; right
W 0 0_2; right
W 3 4; right
Rl 4 0_4; down
W 0_3 0_4;
W 0_2 0_3;
""")
c.draw()
```

[image]

The galvanic isolation is broken by the common connection. This model does not correctly capture the internal currents (see diagrams in my first comment). But would be fine for an ideal case.

A more useful model would be something like this,

```python
c = Circuit("""
Vs 1 0 ac; down
Rs 1 2; right=1.5
R1 2 3; down
L1 3 4 {Lp*N**2/(n**2+1)}; down
R2 4 5; down
L2 5 6 {Lp/(n**2+1)}; down
W 0 6; right
W 4 7; right
Rl 7 8; down
W 6 8; right
K1 L1 L2 1; ignore
""")
c.draw()
```
[image]

Here `Lp` is the inductance of the primary winding and `n` is the step-down ratio, identical to the isolation transformer case. In an ideal (auto)transformer the winding resistances `R1` and `R2` are zero. I have included the here to try an estimate the losses (analytically).

mph-, 2022-02-03:
If we added an autotransformer component, the question is which parameters it should have.  I think the component should ignore winding resistances and core losses.   I guess the three main options are:
1. primary magnetising inductance, secondary magnetising inductance,  mutual inductance
2. primary magnetising inductance, secondary magnetising inductance, coupling coefficient
3. primary leakage inductance, secondary leakage inductance, mutual inductance

What do you think?

danieljfarrell, 2022-02-07:
Yeah it's really tricky making these API decisions because we don't really know how it will be used.

For my case I'm interested in modelling resistive losses in the windings, but someone else might be more interested in bandwidth etc.

If there is doubt I suggest being as consistent as possible with the existing lcapy API.

Above, I parameterised in terms of the magnetising inductance of the primary and the required step-down ratio. 

Note that `L_2` is the inductance of the secondary, but the inductance of the primary is,

```
L_p = L_1 + L_2     (1)
```

You arrive at the two winding inductances by solving (1) and (2).

```
N^2 = L_1 / L_2     (2)
```

I thought this was fairly consistent with lcapy, as isolation transformers are already parameterised with the step-down ratio.

If coupling coefficient can have a default of 1, but could be changes if needed then that would be useful.

So what do I think?

A black box autotransformer could basically reuse the same model as the isolation transformer, but with common connection between the winding. 

So just reuse that, one small modification, and update the symbol?

Or some sensible parameterisation involving inductances.

Maybe using `L_p`, `n` and `K=1` (default value).

Or `L_p` and `L_s` and `K=1` (default value). And we let the user calculate the step-down/up ratio themselves.

These are the magnetising inductances (inductance measured with the opposite winding open).

mph-, 2022-02-08:
Aren't you making the assumption that the reluctance is the same for the two coils when you say that

`N^2 = L_1 / L_2 `

I think whatever parameters we decide for an autotransformer, they should be consistent for the parameters of a transformer. 

My current preference is L1, L2, M where M is the mutual inductance.

If we name the component ATF, what should a transformer be named?  At the moment, to be consistent with SPICE, the TF component refers to an ideal transformer (it even works at DC) and it has a single turns-ratio argument which I always get the wrong way around.

We could have a nonideal keyword, for example,

`TF1 3 2 1 0 nonideal L1 L2 M`
 
but a non-ideal transformer model usually includes winding resistance and core loss.  I guess the losses can be optional parameters that default to zero, say for example:

`TF1 3 2 1 0 nonideal L1 L2 M R1 R2 Rc`

So for symmetry, an autotransformer might be specified with:

`ATF1 2 1 0 nonideal L1 L2 M R1 R2 Rc`