mph-/lcapy #67: Drawing circuits with K netlist statements

danieljfarrell, 2022-01-31:
The following circuit _without_ mutual inductance statement draws fine (in Jupyter notebook)

```python
c = Circuit("""
Vs 1 0 ac; down
Rs 1 2; right=1.5
R1 2 3; down
L1 3 4; down
R2 4 5; down
L2 5 6; down
W 0 6; right
W 4 7; right
Rl 7 8; down
W 6 8; right
""")
```

Adding `K` statement line prevents the `draw` function from finishing. It seems to get struct trying to calculate the layout.

 
```python
c = Circuit("""
Vs 1 0 ac; down
Rs 1 2; right=1.5
R1 2 3; down
L1 3 4; down
R2 4 5; down
L2 5 6; down
W 0 6; right
W 4 7; right
Rl 7 8; down
W 6 8; right
K1 L1 L2 1;
""")
```

I'm using lcapy version 1.1.

mph-, 2022-02-01:
Ahh, this is a bug; a weird graph is created for the schematic node positioning.   This is because the K component makes the assumption that the two coils are side by side and so tries to draw them as a transformer.  The easiest thing is to add the ignore keyword

`K1 L1 L2 1; ignore`

Do you have any ideas of how to show the mutual coupling?

danieljfarrell, 2022-02-01:
In LTSpice the K statement just appears as text next to the coupled inductors as text. Which is I don't particularly like, but it is probably the most pragmatic solution because it is not dependent on the physical layout of the inductors. I suppose the only thing I can thing is a Bezier curve passing through both windings, to indicate shared magnetic flux?

mph-, 2022-02-08:
If you can think of a good algorithm given an arbitrary pairing of inductors with different orientations, I could implement it.