mph-/lcapy #68: Jupyter examples fail : AttributeError: type object 'bytes' has no attribute 'name'

phma-hw, 2022-01-31:
Hi, 
I'm facing multiple issues when trying to use lcapy, so I've come down to the basics and try to run the jupyter examples from the repo.

Testing state-space1.ipynb:
cell [6] : `ss.state_equations()`
fails with

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
d:\tools\python39\lib\site-packages\IPython\core\formatters.py in __call__(self, obj)
    343             method = get_real_method(obj, self.print_method)
    344             if method is not None:
--> 345                 return method()
    346             return None
    347         else:

d:\work\git\lcapy\lcapy\expr.py in wrap(*args, **kwargs)
   1002             if not isinstance(ret, sym.Expr):
   1003                 # Wrap list, tuple, etc.
-> 1004                 return expr(ret)
   1005 
   1006             # Wrap the return value

d:\work\git\lcapy\lcapy\expr.py in expr(arg, override, **assumptions)
   3514 
   3515     if not isinstance(arg, str) and hasattr(arg, '__iter__'):
-> 3516         return exprcontainer(arg, **assumptions)
   3517 
   3518     # Don't set rational=True since this will set rational

d:\work\git\lcapy\lcapy\expr.py in exprcontainer(arg, **assumptions)
   3453         return Vector(arg, **assumptions)
   3454 
-> 3455     raise ValueError('Unsupported exprcontainer %s' % arg.__class__.name)
   3456 
   3457 

AttributeError: type object 'bytes' has no attribute 'name'

```

Many other examples fail with the same error.

I am using
- a fresh clone of lcapy from this repo
- Python '3.9.2 (tags/v3.9.2:1a79785, Feb 19 2021, 13:44:55) [MSC v.1928 64 bit (AMD64)]'
- sympy '1.9'

Any ideas on what causes this failure ?

mph-, 2022-02-01:
Thanks for the bug report. 

I think this should be working now.

Do you know of a good way to check notebooks in a test framework?

phma-hw, 2022-02-01:
Hi, I confirm that your last commits fix this. Thanks a lot for the super-fast help!

I'm not aware of any automated tests on jupyter notebooks, however the results of the execution of cells are saved in the original ipynb files. One way could be to have a ipynb file already populated with the "good" known answers, have jupyter run the whole file, and then diff the ipynb file vs the previous one ?

closed 2022-02-01 by phma-hw