mph-/lcapy #69: Error in lcapy\expr.py : name 'Derivative' is not defined

phma-hw, 2022-01-31:
Hi, 
this issue may be related with #68 : some dependency version/conflict on my installation ?

Here is an attempt to analyze a small circuit composed of 4 capacitors and 4 resistors in a matrix arrangement.

```
from lcapy import *
from matplotlib import pyplot as plt
cir = Circuit()

# Create capacitors
for row in range(2):
    for col in range(2):
        cir.add("C%d%d nR%d nC%d 30e-6" % (row, col, row, col))

# create row resistors
for row in range(2):
    cir.add("RR%d nR%d 0 1e6" % (row, row))
# create col resistors
for col in range(2):
    cir.add("RC%d nC%d 0 1e6" % (col, col))

# Current source in parallel with C00
cir.add("I1 nR0 nC0 {1e-6*rect(t)} ")

cir.I1.i.plot((0, 2))
cir.C00.v.plot((0, 2))

```

The last line fails with
`RuntimeError: Cannot evaluate expression rect(t)/3600 + Derivative(rect(t), t)/40: name 'Derivative' is not defined`

Full trace:
```
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
d:\work\git\lcapy\lcapy\expr.py in evaluate_expr(expr, var, arg)
   2088                 # Try to flush out weirdness using first argument
-> 2089                 response = func(arg0)
   2090             except NameError as e:

d:\work\git\lcapy\lcapy\expr.py in func(arg)
   2074                 try:
-> 2075                     result = func1(arg)
   2076                 except ZeroDivisionError:

<lambdifygenerated-44> in _lambdifygenerated(t)
      3   # Derivative
----> 4 (1/3600)*rect(t) + (1/40)*Derivative(rect(t), t))

NameError: name 'Derivative' is not defined

During handling of the above exception, another exception occurred:

RuntimeError                              Traceback (most recent call last)
<ipython-input-28-369d3c270b71> in <module>
     19 
     20 cir.I1.i.plot((0, 2))
---> 21 cir.C00.v.plot((0, 2))
     22 cir.C01.v.plot((0, 2))

d:\work\git\lcapy\lcapy\texpr.py in plot(self, t, **kwargs)
    152 
    153         from .plot import plot_time
--> 154         return plot_time(self, t, **kwargs)
    155 
    156     def response(self, xvector, tvector, method='bilinear'):

d:\work\git\lcapy\lcapy\plot.py in plot_time(obj, t, plot_type, **kwargs)
    532         obj = cls(rest, **obj.assumptions)
    533 
--> 534     v = obj.evaluate(t)
    535 
    536     if plot_type == None:

d:\work\git\lcapy\lcapy\expr.py in evaluate(self, arg)
   2163             pass
   2164 
-> 2165         return evaluate_expr(expr, var, arg)
   2166 
   2167     def has(self, *patterns):

d:\work\git\lcapy\lcapy\expr.py in evaluate_expr(expr, var, arg)
   2089                 response = func(arg0)
   2090             except NameError as e:
-> 2091                 raise RuntimeError('Cannot evaluate expression %s: %s' % (self, e))
   2092             except AttributeError as e:
   2093                 # Could return NaN but this would take some jiggery

RuntimeError: Cannot evaluate expression rect(t)/3600 + Derivative(rect(t), t)/40: name 'Derivative' is not defined

```

It looks that there could be an import issue in expr.py, as Derivative is a class from sympy ?

mph-, 2022-02-01:
Thanks for the bug report.  

I will have a ponder how to best deal with this tomorrow; there is a problem in the Laplace transform of the rect.

However, with the bug fixed you might not get the answer that you expect,  since Lcapy solves your problem in the Laplace domain but the rect function is ignored for t < 0.  I guess that's why I've never tested sources with a rect function.

mph-, 2022-02-01:
This should be fixed now.  I decided to issue a warning when taking a Laplace transform of the rect function.

phma-hw, 2022-02-02:
Hi, I confirm your last commit fixes it. Many thanks ! (and bigger thanks for the amazing work on lcapy that's a wonderfully useful package)

closed 2022-02-02 by phma-hw