mph-/lcapy #73: Can we add the possibility to selectively choose the StateSpace outputs?

joaoantoniocardoso, 2022-02-13:
Hi, it would be nice to have a way to select which of the outputs to keep, or a way to delete the unwanted ones. For now, I am doing it like so:

```Python
import lcapy as lca

def select_outputs(ss: lca.StateSpace, outputs: list) -> lca.StateSpace:
    outputs = set(outputs)
    outputs_to_delete = set(range(ss.y.rows - 1))
    [
        outputs_to_delete.remove(output)
        for output in outputs
        if output in outputs_to_delete
    ]

    for i, _ in sorted(enumerate(ss.y), reverse=True):
        if i in outputs_to_delete:
            ss.y.row_del(i)
            ss.C.row_del(i)
            ss.D.row_del(i)
    return ss
```

The output:

[image]

Thanks

mph-, 2022-02-14:
Have you tried the `.reduce()` method that will remove selected states?  I wonder if this will help.

joaoantoniocardoso, 2022-02-15:
Hmm, but I want to remove outputs (y), not states (x). But because of your answer I found the method SpaceState.from_circiut, which has parameters to choose the outputs:

```Python
lca.StateSpace.from_circuit(cct_on, node_voltages=['o'], branch_currents=['V1', 'Ro'])
```

For now it is 'solved', but I still think it could be a useful thing to be able to do _after_ the StateSpace is created.

mph-, 2022-02-15:
Okay, how about something like
```diff --git a/lcapy/statespacebase.py b/lcapy/statespacebase.py
index e96b0b82..d77f73fd 100644
--- a/lcapy/statespacebase.py
+++ b/lcapy/statespacebase.py
@@ -592,6 +592,29 @@ class StateSpaceBase(object):
         elim_states = hsv.numpy.squeeze() < threshold
         return self.reduce(elim_states, method)
 
+    def remove_outputs(self, elim_outputs):
+        """Remove outputs given array `elim_outputs` of outputs to remove."""
+
+        from numpy import arange, array, bool_
+
+        if isinstance(elim_outputs[0], (bool, bool_)):
+            elim_outputs = arange(self.Ny)[elim_outputs]
+
+        # SymPy fancy indexing requires lists not ndarrays.
+        mkeep = []
+
+        for i in range(0, self.Ny):
+            if i not in elim_outputs:
+                mkeep.append(i)
+
+        C = self.C[mkeep, :]
+        D = self.D[mkeep, :]
+        y = self._y
+        if y is not None:
+            y = array(y)[mkeep]
+
+        return self.__class__(self.A, self.B, C, D, self.u, y, self.x, self.x0)
+
     def subs(self, *args, **kwargs):
         """Return new state space object after substituting variables in A, B,
         C, D matrices, see sympy.subs for usage.
```

joaoantoniocardoso, 2022-02-15:
Nice! It looks good to me, but don't you think that it would be more cohesive with the `from_circuit` if we pass a list with the output names to remove, instead of a list of booleans?

---

Super related to this - the next few days I'll be facing the trouble of having to model the voltage drop of a diode (I'll consider it a simple Vd+Rd model) and I need the ability to ignore it as a StateSpace input, is it possible? If so, I think it could be an additional parameter `ignore_inputs: List[Str]` on the `from_circuit()`, what do you think?

mph-, 2022-02-15:
That's a better idea.  Are you able to send me a PR?

Adding an `ignore_inputs` argument seems sensible.

joaoantoniocardoso, 2022-02-17:
Yes, I think I can do it next few days, or at least I can draft an initial solution that I'll be using for my thesis.