mph-/lcapy #75: Arrow direction of the voltage source when using the European style

danislab, 2022-02-17:
Hi there, Can we manipulate the arrow direction of the voltage source when we use the European style?

The following code

```python
from lcapy import Circuit
c = Circuit("""W0010 n00 n10; right=1.75
V1 n00 n01 dc 2; down=1.25, i=I_1, v=2V, l=$$
R1 n10 n11 8; down=1.25
W0111 n01 n11; right=1.75
W n01 gnd; down=0.2, ground
""")
c.draw(draw_nodes='connections',style='european')
```

draws the following circuit:

[image]

I would like to have the arrow pointing into the opposite direction, from `n00` to `n01`.
I am using v1.2.3

Thank you!

mph-, 2022-02-17:
Hello

You can set `voltage_dir=EF` (see the circuitikz docs).  Well, you can shortly, since I'm about to push a fix for a typo.

mph-, 2022-02-17:
Sorry, I'm showing my ignorance of the European style.   You just need to draw your component upside down:

`V1 n01 n00 dc 2; up=1.25, i=I_1, v=2V, l=$$`

danislab, 2022-02-21:
Thank you for your replay. The arrow direction of the voltage source for American and European style indeed seems to have caused already many discussions in the Circuitikz community.
If I just draw the components upside down it will also flip the current arrow as shown below
```python
V1 n02 n01 dc 19; up=1.25, i=I_1, v=19V, l=$$
```

[image]

When I draw it upside down and use `voltage_dir=EF` it seems to work.

```python
V1 n02 n01 dc 19; up=1.25, i=I_1, v=19V, voltage_dir=EF, l=$$
```

[image]

Thanks!

closed 2022-02-21 by danislab