mph-/lcapy #76: nodal analysis fails to format properly for a node like 4_1

scottprahl, 2022-04-29:
Consider the twin-T notch filter

```
cct_twin = lcapy.Circuit()
cct_twin.add("""
    Pin 1 0_1; down, v=v_{in}
    W 0_1 0; down=0.2, sground
    W 1 2;
    W 2 2_1; up=1.5
    W 2 2_2; down=1.5
    R1 2_1 3 R; right=1.5, scale=0.75
    R2 3 4_1 R; right=1.5, scale=0.75
    W 4_1 6; down
    C3 3 4 {2*C}; down, scale=0.75
    W 4 0_3; right=0.75
    W 0_3 0; down=0.2, sground
    R3 4 5 {R/2}; down, scale=0.75
    C1 2_2 5 C; scale=0.5
    C2 5 8_1 C; scale=0.5
    W 8_1 6; up
    W 6 7;
    Pout 7 0_2; v=v_{out}
    W 0_2 0; down=0.2, sground
""")
```
[image]

Asking for a nodal analysis reveals a formatting issue

```
na = cct_twin.laplace().nodal_analysis()
na.nodal_equations()
```

One of the nodes analyzed is `4_1 `.  The underscore messes with TeX's mind.  When the two `4_1` entries are changed in the netlist to `41` then everything works fine.

mph-, 2022-05-02:
I've pushed a fix for this.

Do you have any suggestions as to how symbols like these should be formatted?

mph-, 2022-05-03:
Unfortunately, SymPy baulks with a name such as `V1\_2`.

I'll have a go at choosing better nodes; this should be straightforward.

mph-, 2022-05-03:
Ah, the code I had tried to sort the nodes to avoid this sort of thing but it wasn't smart enough.  It should be fixed now.

scottprahl, 2022-05-03:
Great news that the fix for using better nodes was straightforward!  Thanks

Every mini-language has its own ideas about how to handle special characters.  Eliding the `_` is a reasonable solution to this issue.

scottprahl, 2022-05-04:
I pulled the latest and can confirm that this is now fixed.  

Nodal analysis now finds node 6 instead of node 4_1.  

Furthermore, if a node with an underscore is found, then the resulting equation is formatted properly.

closed 2022-05-04 by scottprahl