mph-/lcapy #77: pip install lcapy installs old version of sympy

scottprahl, 2022-04-29:
I had recently updated `sympy` to `1.10.1` because Laplace transform handling is significantly improved (see https://github.com/sympy/sympy/pull/22376) .  After installing `lcapy` my version of `sympy` was downgraded to `1.8.0`

Currently, `lcapy` requires `sympy>=1.7`.  You might consider bumping the requirement in `setup.cfg` to be `>=1.10.1`.  (This will mean bumping python to >=3.8 as well

mph-, 2022-05-02:
Thanks for letting me know.  I'm in the process of making a new release of Lcapy.  

New releases of Sympy usually breaks things.  Have you seen any evidence of this with 1.10.1?

scottprahl, 2022-05-02:
In my experience, I have not seen any evidence of problems using 1.10.1.

Honestly, though, the Laplace transform stuff prior to 1.10 was horrible. I can only image that you needed to make many allowances for suboptimal behavior. 

BTW, if you're creating a new release, you might consider using something like [test_all_notebooks.py](https://github.com/scottprahl/miepython/blob/ee86245489831a9432becddd7697cb9b18586885/tests/test_all_notebooks.py).  I use this to run all the Jupyter notebooks in my projects.  No testing of individual cell outputs is done, but at least I am assured that notebooks run to completion.  (The only changes I have made to the script is to silence some pylint warnings.  There is a link to the original script in the file itself.)

mph-, 2022-05-12:
Yes, the earlier Laplace support in SymPy was poor; that's why I added many Laplace transforms to Lcapy and only used SymPy as a last resort.

My tests with SymPy-1.10.1 seem to be fine so I've added this to the Lcapy-1.5 that I've just released.

closed 2022-05-12 by mph-