mph-/lcapy #78: Feature Request: Automatically drawing ground symbols

scottprahl, 2022-05-02:
So, `lcapy` is great, but manually connecting grounds is a hassle.  I mess them up regularly.

For example, `lcapy` can analyze the following circuit, but `cct.draw()` fails.

```
cct = Circuit()
cct.add("""
    Iin 1 0 {6*u(t)+3*delta(t)}; down, size=1.5
    W 1 2; size=1.5
    R1 2 0 2; down
    R2 2 3 1; size=1.5
    L1 3 0 0.2; down, i=i
""")
```

I'd like is to have an option like `autoground=sground` so that `lcapy` just drew a signal ground symbol whenever a `0` node was encountered.  The default `autoground=None` would give the current behavior.  

```
cct = Circuit()
cct.add("""
    Iin 1 0_1 {6*u(t)+3*delta(t)}; down
    W 0_1 0; down=0.1, sground
    W 1 2; size=1.5
    R1 2 0_2 2; down
    W 0_2 0; down=0.1, sground
    R2 2 3 1; size=1.5
    L1 3 0_3 0.2; down
    W 0_3 0; down=0.1, sground
    ;label_ids=none
""")
cct.draw()

```

[image]

mph-, 2022-05-03:
This is an excellent suggestion.   That should be straightforward to implement.

A few weeks ago I toyed with adding implicit connections to oneport devices such a resistors, capacitors, etc.  This should also help.

scottprahl, 2022-05-03:
Looking forward to testing this out!

mph-, 2022-05-04:
Did this work as you expected?

scottprahl, 2022-05-05:
I tried it, but it did not work and just figured that the code was a not yet complete.  Here is what I see:

* Using `lcapy` commit f66125
* Python 3.9.12
* sympy 1.10.1
* jupyterlab 3.2.5

Testing with

```
cct = Circuit()
cct.add("""
    Iin 1 0 {6*u(t)+3*delta(t)}; down, size=1.5
    W 1 2; size=1.5
    R1 2 0 2; down
    R2 2 3 1; size=1.5
    L1 3 0 0.2; down, i=i
    ;autoground=sground
""")
cct.draw()
```

`RuntimeError: The horizontal schematic graph has a loop.  For example, a node needs to be both above and below another node.  Probably a component is attached to the wrong nodes.
`

> ---------------------------------------------------------------------------
> RecursionError                            Traceback (most recent call last)
> ~/Documents/Code/git/lcapy/lcapy/schemgraph.py in path_to_closest_known(self, from_gnode, forward)
>     705         try:
> --> 706             traverse(from_gnode)
>     707         except RuntimeError:
> 
> ~/Documents/Code/git/lcapy/lcapy/schemgraph.py in traverse(gnode)
>     696                 # TODO, memoize
> --> 697                 dist = traverse(next_gnode) - edge.size
>     698                 if dist < min_dist:
> 
> ... last 1 frames repeated, from the frame below ...
> 
> ~/Documents/Code/git/lcapy/lcapy/schemgraph.py in traverse(gnode)
>     696                 # TODO, memoize
> --> 697                 dist = traverse(next_gnode) - edge.size
>     698                 if dist < min_dist:
> 
> RecursionError: maximum recursion depth exceeded in comparison
> 
> During handling of the above exception, another exception occurred:
> 
> RuntimeError                              Traceback (most recent call last)
> <ipython-input-2-bb1653043a81> in <module>
>       8     ;autoground=sground
>       9 """)
> ---> 10 cct.draw()
> 
> ~/Documents/Code/git/lcapy/lcapy/netlistmixin.py in draw(self, filename, **kwargs)
>    2146             cct = cct.s_model()
>    2147 
> -> 2148         return cct.sch.draw(filename=filename, **kwargs)
>    2149 
>    2150     @property
> 
> ~/Documents/Code/git/lcapy/lcapy/schematic.py in draw(self, filename, **kwargs)
>     619 
>     620             png_filename = tmpfilename('.png')
> --> 621             self.tikz_draw(png_filename, **kwargs)
>     622 
>     623             # Create and display PNG image object.
> 
> ~/Documents/Code/git/lcapy/lcapy/schematic.py in tikz_draw(self, filename, **kwargs)
>     445         latexrunner = LatexRunner(self.debug & 2)
>     446 
> --> 447         content = self._tikz_draw(style_args=style_args, **kwargs)
>     448 
>     449         if nosave:
> 
> ~/Documents/Code/git/lcapy/lcapy/schematic.py in _tikz_draw(self, style_args, **kwargs)
>     343         self._setup()
>     344 
> --> 345         self._positions_calculate(method, self.debug)
>     346 
>     347         # Note, scale does not scale the font size.
> 
> ~/Documents/Code/git/lcapy/lcapy/schematic.py in _positions_calculate(self, method, debug)
>     335         if self.debug & 4:
>     336             print('Solving graphs')
> --> 337         self.width, self.height = placer.solve(self.node_spacing)
>     338 
>     339     def _tikz_draw(self, style_args='', **kwargs):
> 
> ~/Documents/Code/git/lcapy/lcapy/schemplacerbase.py in solve(self, node_spacing)
>      96         self._make_graphs()
>      97 
> ---> 98         xpos, width = self.xgraph.solve()
>      99         ypos, height = self.ygraph.solve()
>     100 
> 
> ~/Documents/Code/git/lcapy/lcapy/schemgraph.py in solve(self, stage)
>     591 
>     592         # Assign remaining gnodes.
> --> 593         self.assign_stretchy(unknown)
>     594 
>     595         if unknown != []:
> 
> ~/Documents/Code/git/lcapy/lcapy/schemgraph.py in assign_stretchy(self, unknown)
>     411                     break
>     412 
> --> 413                 changes = self.assign_stretchy1(gnode, unknown)
>     414                 if changes:
>     415                     self.assign_fixed(unknown)
> 
> ~/Documents/Code/git/lcapy/lcapy/schemgraph.py in assign_stretchy1(self, gnode, unknown)
>     330         # a stretchy path of a longer distance than a fixed path.
>     331 
> --> 332         to_path = self.path_to_closest_known(gnode, forward=True)
>     333         from_path = self.path_to_closest_known(gnode, forward=False)
>     334 
> 
> ~/Documents/Code/git/lcapy/lcapy/schemgraph.py in path_to_closest_known(self, from_gnode, forward)
>     713             if self.debug:
>     714                 msg += str(self)
> --> 715             raise RuntimeError(msg)
>     716 
>     717         return self.makepath(from_gnode)
> 
> RuntimeError: The horizontal schematic graph has a loop.  For example, a node needs to be both above and below another node.  Probably a component is attached to the wrong nodes.

mph-, 2022-05-05:
Oh, I see.

It works with

    Iin 1 0_1 {6*u(t)+3*delta(t)}; down
    W 1 2; size=1.5
    R1 2 0_2 2; down
    R2 2 3 1; size=1.5
    L1 3 0_3 0.2; down
    ;label_ids=none, autoground=sground

but I will need to automatically rename all the ground nodes to work with your example.

scottprahl, 2022-05-06:
**It now works!  It looks great.**  

Two minor comments.

1. Consider modifying the API so users don't need to remember the names of the grounds (unless they really care :-)

`autoground=False`; default
`autoground=True`;  would imply `autoground=ground` or `autoground=sground`

2. This twin-T filter is perfect except for the missing circles on the ground nodes of the ports

```
cct_twin = Circuit()
cct_twin.add("""
    Pin 1 0; down, v=v_{in}
    W 1 2;
    W 2_1 2; down
    W 2 2_2; down
    R1 2_1 3 R; right=1.5
    R2 3 4_1 R; right=1.5
    W 4_1 6; down
    C3 3 4 {2*C}; down
    W 4 0; right=0.75
    R3 4 5 {R/2}; down
    C1 2_2 5 C;
    C2 5 8_1 C;
    W 6 8_1; down
    W 6 7;
    Pout 7 0; v=v_{out}
    ;draw_nodes=connections, label_ids=False, autoground=sground
""")
cct_twin.draw()
```

[image]

mph-, 2022-05-06:
I've pushed a fix for this; it was surprisingly tricky.  I now need to check for unintended side-effects.

scottprahl, 2022-05-07:
Tested both requests: Port terminals now are correct and autoground=True works fine.

I think this is a good feature for the next release.  Thanks!

closed 2022-05-07 by scottprahl

mph-, 2022-05-07:
Let me know if you have any more good ideas.

BTW, you can now do

    Iin 1 0 {6*u(t)+3*delta(t)}; down, sground
    W 1 2; size=1.5
    R1 2 0 2; down, sground
    R2 2 3 1; size=1.5
    L1 3 0 0.2; down, sground
    ;label_ids=none

which gives

[image]