mph-/lcapy #80: node names and current name overlap

lucatatas, 2022-05-16:
Hi, 
when I want to draw a circuit, sometime the name of a special node and the name of a current in the circuit overlap which results in the problem that neither of the names is readable.

I have attached some screenshots and some example code:
`from lcapy import Circuit
c = Circuit("""
V1 n10 n00 dc 12; left=1.75, i=I_{V1}, v=12V, l=$$, voltage dir=EF
W0001 n00 n01; down=1.25\nW1020 n10 n20; right=1.75
R1 n01 n02 7; down=1.25, i=I_R_1
W0212 n02 n12; right=1.75
W1112 n11 n12; down=1.25
W2021 n20 n21; down=1.25
D1 n11 n21; right=1.75, scale=0.75, i=I_D_1
W n10 gnd; down=0.2, ground
""")
c.draw(draw_nodes='connections',label_nodes="n00,n10,n02",style="european")`

[image]
[image]
[image]
[image]

Thanks in advance!

mph-, 2022-05-20:
This is a tricky problem to automatically solve.

The easiest solution is to increase the component size, say `R1 1 2; right=1.5`

Alternatively, you can move the current label, see https://lcapy.readthedocs.io/en/latest/schematics.html?highlight=current%20label#current-and-flow-labels

I welcome any suggestions you might have to improve this.

mph-, 2022-05-27:
I have added a schematic attribute `node_label_anchor` that will customize where the node labels are placed.  This will not fix your problem in general but can help.

For example,

    R 1 2; right
    C 2 3; right

gives

[image]

and

    R 1 2; right
    C 2 3; right
    ; node_label_anchor=south west

gives

[image]

mph-, 2022-06-09:
Have my suggestions helped?  If not, I will reopen the issue.

closed 2022-06-09 by mph-