mph-/lcapy #84: Nodal analysis of Laplace model does not include initial conditions

scottprahl, 2022-06-14:
Consider the following circuit with initial conditions

```
import lcapy
cct=lcapy.Circuit("""
C 1 0 1 15; down
L 1 2 1 5; right
R1 2 0 3; down
R2 2 3 15; right
V 3 0 15; down
;autoground=True,label_ids=False
""")
cct.s_model().draw()
```

which shows the correct circuit with voltage sources to account for the initial conditions.

[image]

Unfortunately `nodal_analysis()` fails to include these voltage sources

```
n = cct.laplace().nodal_analysis()
n.nodal_equations()
```

[image]

`lcapy` *does include* the initial conditions when analyzing the circuit (i.e., `cct[2].v` gives the correct answer).  So maybe this is just overlooked in `nodal_analysis()`?

mph-, 2022-06-16:
Thanks for the example.  Yes, this was overlooked.  The same problem applies for loop analysis.  It should be straightforward to fix.   Stay tuned once I've had a sleep...

mph-, 2022-06-16:
I think I have fixed this but will need to write some unit tests.

scottprahl, 2022-06-16:
Things aren't quite right yet.  Using version 7ca1151e gives

[image]

The correct equations should be

```
v1, v2, s, t = sympy.symbols('v_1 v_2 s t')
eq1 = (v1-15/s)/(1/s) + (v1-(v2-5))/s
eq2 = -(v1-(v2-5))/s + (v2-15/s)/15 + v2/3
sympy.solve([eq1,eq2], (v1,v2))
```

(the inverse Laplace transform gives the same solutions as `cct[1].v` and `cct[2].v`)

It looks like the initial condition for the capacitor voltage is off by $s^2$

mph-, 2022-06-17:
Sorry, in my haste I multiplied by s rather than dividing by s.

scottprahl, 2022-06-17:
It works now!  Thanks for the fix.

closed 2022-06-17 by scottprahl