mph-/lcapy #85: Solving equations generated by loop or mesh analysis

scottprahl, 2022-06-17:
The docs for `Circuit Analysis > Linear Circuit Analysis > Mesh Analysis` and `Circuit Analysis > Linear Circuit Analysis > Nodal Analysis` have nice examples of generating the nodal and mesh equations.

Unfortunately, the next step, a straight-forward method for solving the system is not given.

To be specific, consider the circuit

```
cct=lcapy.Circuit("""
C 1 0 1 15; down
L 1 2 1 5; right
R1 2 0 3; down
R2 2 3 15; right
V 3 0 15; down
;autoground=True,label_ids=False""")
n = cct.laplace().nodal_analysis()
eqn_dict = n.nodal_equations()
```

Extracting the list of equations from the `eqn_dict` took a bit work to figure out, but this is what I ended up with

```
eqn_list = list(eqn_dict.sympy.values())
eqn_list
```

[image]

Because the voltages are written as $V_1(s)$, the following fails

```
from sympy import Eq
V1, V2, V3, s = sympy.symbols('V_1 V_2 V_3 s')
sympy.solve(eqn_list, (V1,V2,V3))
```

Currently, I `print(eqns)` and manually delete all the '(s)' entries and then do something like

```
V1, V2, V3, s = sympy.symbols('V_1 V_2 V_3 s')
sympy.solve([s*(V1 - 15/s) + (V1 - V2 + 5)/s, 
             2*V2/5 - V3/15 + (-V1 + V2 + 5)/s, 
             V3 - 15/s],
            (V1,V2,V3))
```

So, what options are there?  Hopefully, I am overlooking the supported method.  If so, then please document it!

If not, then ideally `.solve()` would be overloaded on `ExprDict` so that the following works (but I expect that this is hard)

```
n.nodal_equations().solve((V1,V2,V3))
```

Alternatively, you could modify `eqn_dict.sympy` to omit the '(s)' from the returned equations --- or create a new version `eqn_dict.sympy_nofunc` that achieves the same thing.

mph-, 2022-06-20:
Hmmm, the `solve_laplace()` method used to work.  I guess SymPy changed behind my back.

Anyway, I've made the `solve()` methods more robust and you can just use `n.solve_laplace()`.

mph-, 2022-06-20:
Yes, this was easy to add.  However, I haven't tested it yet.  I've got a networkx/pandas/numpy  bug to resolve first.

closed 2022-06-21 by mph-