mph-/lcapy #86: Autoground fails with op-amps

scottprahl, 2022-06-19:
The inverting amplifier that does not use autoground.
```
import lcapy
a = lcapy.Circuit("""
    E 1 0 opamp 3 2 Ad; right
    W 2 2_1; down
    R1 2_1 0_2 R1; down
    R2 2_1 1_1 R2; right
    W 1 1_1; down
    W 3_1 3_2; down
    Vs 3_2 0_3 Vs; down
    W 0_3 0_1; down
    W 3_1 3_3; right
    W 3_3 3; right
    W 1 12; right
    P 12 0; down
    W 0_1 0_2; right
    W 0_2 0; right
    ; label_nodes=True
""")
a[12].v
```
[image]

Consider the same circuit without separate ground nodes and with `autoground=True`. The following does not work .
```
import lcapy
a = lcapy.Circuit("""
    E 1 0 opamp 3 2 Ad; right
    W 2 2_1; down
    R1 2_1 0 R1; down
    R2 2_1 1_1 R2; right
    W 1 1_1; down
    W 3_1 3_2; down
    Vs 3_2 0 Vs; down
    W 3_1 3_3; right
    W 3_3 3; right
    W 1 12; right
    P 12 0; down
    ; label_nodes=True; autoground=True
""")
a.draw()
```

I suspect that op-amps were just overlooked when implementing `autoground`.  

The following does at least draw the circuit,

```
a = lcapy.Circuit("""
    E 1 0_1 opamp 3 2 Ad; right
    W 2 2_1; down
    R1 2_1 0 R1; down
    R2 2_1 1_1 R2; right
    W 1 1_1; down
    W 3_1 3_2; down
    Vs 3_2 0 Vs; down
#    W 0_3 0_1; down
    W 3_1 3_3; right
    W 3_3 3; right
    W 1 12; right
    P 12 0; down
#    W 0_1 0_2; right
#    W 0_2 0; right
    ; label_nodes=True, autoground=True
""")
a.draw()
```

[image]

but, of course, `a[12].v` fails.

mph-, 2022-06-20:
Thanks for the testcase.  I've pushed a fix.

Note, there is a syntax error in your example.

    ; draw_nodes=True; autoground=True

should be

    ; draw_nodes=True, autoground=True

I'll see if I can make the parser more accommodating but am worried by unexpected consequences.

scottprahl, 2022-06-20:
Confirmed that it works!

Oddly the semicolon error arose through copy-and-paste between Jupyter and the bug report :-)

Probably not worth making the parser more permissive :-)

closed 2022-06-20 by scottprahl