mph-/lcapy #89: Unable to draw a circuit or schematic

ashishmoudghil, 2022-10-12:
Hi 

i tried many thing but i am not able to get any thing to draw

from lcapy import Circuit
cct = Circuit("""
V 1 0 {V(s)}; down
R 1 2; right
C 2 _0_2; down
W 0 _0_2; right""")
tried 
cct.draw()
or
cct.draw('voltage-divider.pdf')
or
cct.draw(svg=True)
or
cct.draw(png=True)

even tried adding %matplotlib inline

in the terminal pdflatex (gives)
This is pdfTeX, Version 3.1415926-2.5-1.40.14 (TeX Live 2013)
 restricted \write18 enabled.

can you please help me resolve this?

mph-, 2022-10-13:
Can you send me the output from `cct.draw(debug=2) `?

ashishmoudghil, 2022-10-13:
Chdir: /tmp
Running: pdflatex -interaction batchmode tmppoph0d72.tex
Chdir: /home/aaaaaa/Dashboard
Removing: /tmp/tmppoph0d72.tex
Removing: /tmp/tmppoph0d72.log

mph-, 2022-10-13:
That's bizarre.  It should then try to generate a png file.

Unfortunately, I cannot replicate this on my computer.  Do you know how to use the Python debugger?

Can you also try `cct.draw('foo.png', debug=2)` ?

haulisson, 2023-05-09:
Hello everyone, I am writing to report an issue with schematic with **nfetd** and **pfetd** in the example. Has anyone else experienced the same problem or knows how to fix it? Thank you in advance for your help?

I using the colab with circuitikz-1.4.6.sty

cct = Circuit("""
M5 19 20 21  M5; up, kind=nfetd, l=nfetd
M6 22 23 21  M6; up, kind=pfetd, l=pfetd
M7 22 24 25  M7; up, kind=nfet, l=nfet
M8 26 27 25  M8; up, kind=pfet, l=pfet
M9 26 28 29  M9; up, kind=nigfetd, l=nigfetd# Hack to include labels in bounding box
O 19 18; up=0.8
O 28 30; down=0.4
; label_nodes=none, draw_nodes=connections
""")
cct.draw()

Result
https://academicoifrnedu-my.sharepoint.com/:i:/g/personal/haulisson_jody_academico_ifrn_edu_br/ERXFOD2an9RCt539Ob4M1e8BPQaCjX50R4rjAcHNYD60qQ?e=oZgtnR

mph-, 2023-05-21:
I have no idea of what is happening.  I get this image with  Circuitikz  1.5.5

[image]

Can you create a simpler test case?