mph-/lcapy #90: voltage symbol reversed

nkrstajic, 2022-10-17:
Hi,

Enjoying using lcapy and re-learning circuits along the way! Thanks for all the effort.

I've noticed a possible bug in how dc voltage is displayed. For example, for lcapy 1.8 I run the code example below:

```
from lcapy import Circuit
 
cct = Circuit()
cct.add("""
Vi 1 0_1 step 20; down
C 1 2; right, size=1.5
R 2 0; down
W 0_1 0; right
W 0 0_2; right, size=0.5
P1 2_2 0_2; down
W 2 2_2;right, size=0.5""")
 
cct.draw(style='american')
```

and get a circuit

[image]

The plus and minus on voltage symbol seem reversed - or am I getting something wrong? There are jupyter noteboks showing plus sign on top but when I run it locally, it is reversed.

Thanks

Nikola

mph-, 2022-10-18:
That is peculiar.  From IPython I get the positive sign on top.

I wonder if you have an old version of circuitikz installed?

Can you try `cct.draw(debug=1)`?  This will display the circuitikz version.

nkrstajic, 2022-10-20:
Yes, my circuitkz was old and could not update it on my work computer (which limits admin and update rights). I've tested it now separately and works fine. Thanks, circuit schematics look really good and symbolic approach is new to me ...

mph-, 2022-10-20:
Okay.  You could install circuitikz on your TEXINPUTS path for your work computer as a work-around.

closed 2022-10-20 by mph-