mph-/lcapy #92: What's wrong with this AC simulation?

oe8bck, 2022-10-27:
I would like to get the transfer function of this circuit:

```
from lcapy import Circuit
cct = Circuit("""
Vin 3 0 ac 1; down=1.5,l=V_{in}
W 0 0_0; right
W 3_3 3; left
A1p 3; l=vin, anchor=south
R2 3_3 8_1; right,l^=R_2
A2p 8_1; l=vgnd, anchor=west
W 8_1 8_2; up
W 10_0 10_1; up
R4 8_2 10_1; right,l=R_4
W 8_2 8_3; up
C3 8_3 12_; right,l=C_3
W 10_1 10_2; right
W 10_2 10_3; up
R3 12_ 10_3; right,l=R_3
E1 10_0 0_0 10_0 0_0 1e6; down, l=A \cdot vgnd
W 0_0 0_1; right=1.5
W 0_0 0_2; down=0.1,implicit, l=V_{CM}
W 10_2 10; right=0.5
A 10; l=vout, anchor=west
;help_lines=0,label_nodes=none,scale=0.4
""")

cct.draw() #style='european')
cct3[10].v.pprint()
```

But the node voltage at node 10 is 0V. How to properly give the gain stage E1 a gain in order to get the AC transfer function? The expected behavkiour is a pole and a zero, because at low frequencies, only R4 is in the feedback whereas at high frequencies also R3 is in parallel.

Many thanks for your answers in advance!
KR,
Christof

mph-, 2022-10-27:
Hello Christof

The problem is that the controlling nodes for E1 are the same as the output nodes.  Instead you need, 

    E1 10_0 0_0 8_1 0_0 1e6; down, l=A \cdot vgnd

I also suggest that you use `A` for 1e6.  You can then consider the result in the limit when `A` tends to infinity.

oe8bck, 2022-10-28:
Thanks a lot for the fast answer! As usual, PEBKAC ;-)

closed 2022-10-30 by mph-